prusa3d/PrusaSlicer #69: Segmentation Fault at startup

cubiq, 2016-12-20:
### Version
1.31.6

### Operating system type + version
Arch Linux

### Behavior
Application crashes on startup with Segmentation Fault. I believe it is related to this https://github.com/alexrj/Slic3r/issues/3540

Basically an issue introduced by opengl 0.70

Debug below
```
Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1009,wx containers,compatible with 2.8),
and your program used 3.0 (wchar_t,compiler with C++ ABI 1010,wx containers,compatible with 2.8).
Signal SEGV at /usr/share/perl5/vendor_perl/Slic3r/GUI/3DScene.pm line 138.
        Slic3r::GUI::3DScene::Base::__ANON__[/usr/share/perl5/vendor_perl/Slic3r/GUI/3DScene.pm:139](Slic3r::GUI::Plater::3D=HASH(0x4915a38), Wx::PaintEvent=SCALAR(0x4c73ac0)) called at /usr/share/perl5/vendor_perl/Slic3r/GUI/MainFrame.pm line 73
        eval {...} called at /usr/share/perl5/vendor_perl/Slic3r/GUI/MainFrame.pm line 73
        Slic3r::GUI::MainFrame::new("Slic3r::GUI::MainFrame", "mode", "expert", "no_controller", "", "no_plater", undef) called at /usr/share/perl5/vendor_perl/Slic3r/GUI.pm line 143
        Slic3r::GUI::OnInit(Slic3r::GUI=HASH(0x2669fc8)) called at /usr/lib/perl5/vendor_perl/Wx/App.pm line 36
        eval {...} called at /usr/lib/perl5/vendor_perl/Wx/App.pm line 36
        Wx::App::new("Slic3r::GUI") called at /usr/bin/vendor_perl/slic3r line 110
Aborted (core dumped)
```

bubnikv, 2016-12-20:
Matteo,

Thanks for heads up. I am aware of the Perl::OpenGL issue and at least for the build system we are using I solved the issue by pulling the older OpenGL library version. You seem to either using the new one, or having both installed.

@alexrj converted his Build.PL to install the perl libraries locally. I will likely do the same in the future, which will these library incompatibility issues. Ideally one would rewrite as much as possible of the OpenGL stuff from Perl to C++.

CReimer, 2016-12-21:
@alexrj merged pull request https://github.com/alexrj/Slic3r/pull/3575. Which fixes this problem altogether.

bubnikv, 2017-01-20:
Hopefully fixed by 
https://github.com/prusa3d/Slic3r/commit/c0b3de6248f6e6d50624b3d2132c823fb65af95e

closed 2017-01-20 by bubnikv