prusa3d/PrusaSlicer #70: High risk of crash when rotating while in Layers tab

DrLex0, 2016-12-20:
### Version
1.31.6-prusa3d

### Operating system type + version
Mac OS X 10.11.6

### Behavior
* Slic3r has a high risk of crashing when rotating a model while the Layers tab is active.
* Steps to reproduce:
  * Load the STL file and settings. It seems that any model significantly more complex than a simple cube, and any settings, will trigger the problem.
  * Go to the Layers tab
  * Hit the 45° ccw or 45° cw buttons repeatedly, wait every time for background processing to finish.
* _Expected Results_
  * I can keep rotating the object indefinitely.
* _Actual Results_
  * At the moment I hit the rotate button, there is a risk that Slic3r crashes to desktop. The risk could be anything between 10% and 50%, it is too random to get a decent estimate. Sometimes I can rotate 20 times in a row, sometimes it goes boom the first time I hit the button. The problem does not occur in the 3D, 2D, or Preview tabs.

Comparing with 1.2.9 is pointless because that one crashes randomly when doing pretty much anything.

#### STL/Config (.ZIP) where problem occurs
[ConfigAndSTL.zip](https://github.com/prusa3d/Slic3r/files/664193/ConfigAndSTL.zip)

[CrashReport.txt](https://github.com/prusa3d/Slic3r/files/664205/CrashReport.txt)

bubnikv, 2016-12-22:
I am afraid there is not much to do right now. The background processing is broken and you are better to disable it and start slicing manually.

bubnikv, 2017-03-23:
I think this is due to the background processing enabled. We have the background processing disabled now by default. Closing.

closed 2017-03-23 by bubnikv