prusa3d/PrusaSlicer #71: Caught C++ exception of unknown type

DrLex0, 2016-12-20:
### Version
1.31.6-prusa3d

### Operating system type + version
Mac OS X 10.11.6

### Behavior
* Slicing a particular model causes multiple dialog boxes with “exception of unknown type” to pop up, and Slic3r freezes when trying to do anything after that.
* _Steps needed to reproduce the problem_
  * Load the included settings, and the STL file
* _Expected Results_
  * It slices normally
* _Actual Results_
  * Multiple dialog boxes pop up, stating: *“Thread 7 terminated abnormally: Caught C++ exception of unknown type at /Applications/Slic3r.app/Contents/lib/site_perl/5.22.0/Slic3r/Print/Object.pm line 468”.* After clicking them away and trying anything in Slic3r, it freezes with the spinning beachball of death.

After making some modifications to the model, the crash went away. For instance the newer model from issue #70 does not cause the crash. Rotating the model in any direction also avoids the crash, it only seems to happen in this particular pose.

#### STL/Config (.ZIP) where problem occurs
[Issue71.zip](https://github.com/prusa3d/Slic3r/files/664324/Issue71.zip)

bubnikv, 2016-12-22:
I tested with Windows 10 64bit and I could not reproduce the crash you reported. Maybe it is connected to the background processing? Could you please try to reproduce the crash with the background processing disabled?

We shall disable the background processing by default for our builds. I will do that for the next official build.

DrLex0, 2016-12-28:
When background processing is disabled, I get the same error dialogs at the moment I push the ‘Slice now’ button, or try to export the G-Code file. I have also only seen this with this particular STL file in this particular orientation.

bubnikv, 2017-03-23:
I was not able to reproduce the crashes and we are now couple of versions up. Closing. If you encounter such crashes, please open a new issue.

closed 2017-03-23 by bubnikv