prusa3d/PrusaSlicer #72: Compilation issue with recent code base

Sebastianv650, 2016-12-21:
### Version
just from today

### Operating system type + version
Win10 64bit

### Behavior
I tried to compile the latest code base, but something goes wrong during perl build.pl. It's always complaining about a missing subroutine:

[image]

CReimer, 2016-12-21:
That's a duplicate of #52
The perl Module::Build system doesn't seem to detect that it actually links against an outdated object file.

Clean up your source tree an build again. That will fix this.

Sebastianv650, 2016-12-21:
Oh, I wasn't recognising this..
I'm using perl only for Slic3r, and all I know is how to use it for building it. Can you give me a hint where and how to start this clean up? The Slic3r-Prusa directory is a fresh one, no old Slic3r files are overwritten or reused therefore.

playlet, 2016-12-21:
I can't compile either. Cloned the directory a few minutes ago and it fails leaving me with this. Is it me, or is this not yet compileable? Win10x64

[image]

bubnikv, 2017-01-20:
Does it compile with the current master?
Does it compile when linking boost statically? Please execute following line before running perl Build.PL
set SLIC3R_STATIC=1

Sebastianv650, 2017-01-20:
Yes, it's now compiling with this option :-)

bubnikv, 2017-01-20:
great, thanks.

playlet, 2017-01-20:
Now it works for me, too. Thank you very much sir! :)

bubnikv, 2017-02-02:
Thanks for a feedback. I shall fix the compilation with dynamically linked boost then.

bubnikv, 2017-02-15:
Fixed the title to Slic3r does not compile with dynamic boost

bubnikv, 2017-03-23:
Slic3r seems to be linking against a dynamic boost library on Linux just fine.
On Windows, we recommend to use Visual Studio Community 2013 with our pre-compiled perl and libraries. Slic3r Prusa Eiditon is pre-compiled to link statically on Windows and the static pre-compiled libraries are provided.
https://github.com/prusa3d/Slic3r/wiki/How-to-compile-Slic3r-Prusa-Edition-on-MS-Windows

closed 2017-03-23 by bubnikv