prusa3d/PrusaSlicer #74: Seam position "aligned" does not work

burbilog, 2016-12-23:
### Version
1.31.6-prusa3d-linux64

### Operating system type + version
Ubuntu 14.04.05

### Behavior

Problem: if I choose seam position "aligned" it does not align all seams like original slic3r does, but instead it works exactly the same as "nearest" option. It's easy to reproduce, just select "aligned" setting and print. Current original slic3r with the same "aligned" setting and the same config prints these seams correctly.

#### STL/Config (.ZIP) where problem occurs

Weird. Can't upload zip file. Anyway, it seems to happens with any config settings, just try to turn on "aligned" option.

bubnikv, 2016-12-24:
I have to look into that, but the issue is deeper. The "aligned" seam never worked, the code keeps only a single "seam" value per object slice, so if you had multiple islands, they were only aligned by a pure luck (the strategy "take the closest point to the last print head position often worked". So the "aligned" seem was really mostly useful for the vase prints IMHO.

I personally think it that Slic3r excels in the automatic placement of the seam and a good extension would to paint the seam line on the object where the seam cannot be hidden (usually on round objects). Where there are concave or convex corners, one is best to let Slic3r put the seams there. Only if there is no such spot on the object, then some form of manual intervention is needed.

burbilog, 2016-12-24:
> The "aligned" seam never worked, the code keeps only a single "seam" value per object slice, so if you had multiple islands, they were only aligned by a pure luck (the strategy "take the closest [...]

Eh... no. Not only for the vase, but for some other objects too. Like simple geometric objects, for example. I can't insert photos right now because I left all these 20 mm test cubes at work, but cubes with "aligned" options are really smooth while cubes with "nearest" have regular saw-like pattern on every vertical edge (and it shows on flat sides too)... yes, mechanics on my $200 printer kit is not that great, but hey, "aligned" worked for  me!  And even if some option works *sometimes* it's good to have it than to have nothing...

> I personally think it that Slic3r excels in the automatic placement of the seam

Well, apparently, sometimes it does not.  And since the seam can't be completely eliminated, may be it's better to implement another feature from S3D -- it has an option to specify x and y point for the seam, so it tries to place the seam closest to that point.

tcm0116, 2017-02-15:
I'm having this issue with standard calibration cubes on 1.33.3.

I spent several hours trying to figure out why I was getting a sawtooth pattern on my cubes, when I finally realized that: 1) my printer is over-extruding when unretracting, and 2) Slic3r is not aligning the seams.

With a 2x2x1 cm cube, every layer starts at a different corner. Using the "XYZ Calibration Cube" from Thingiverse (http://www.thingiverse.com/thing:1278865), the portions of the cube above and below the letters are not aligned, but the portion in the middle with the letters is aligned.

bubnikv, 2017-02-22:
Hopefully fixed by https://github.com/prusa3d/Slic3r/commit/82aed45816b300dc3a8b391c866a573cd979fcbc

The way it works now is following:
Slic3r will add some negative penalty to all perimeter points near the last seam.
Once the perimeter point with minimum penalty is found, its penalty is compared
to a point closest to the last seam. If the penalty of the point 
closest to the last seam is nearly as good as the minimum penalty,
the point closest to the last seam is picked instead. This heuristics
will hide the seams into corners if possible, but if not possible, it will
strive to align the seams precisely.

Closing the issue, please feel free to reopen the issue, if it does not work as expected.

closed 2017-02-22 by bubnikv