prusa3d/PrusaSlicer #75: Stuck on "Processing triangulated mesh..." when i add an STL file

myckeul, 2016-12-27:
### Version
1.31.6

### Operating system type + version
Ubuntu Linux 16.04.1 LTS Xenial 

### Behavior
When i try to add an STL file, Slic3r stuck on "Processing triangulated mesh..." and i can't do anything more.
This happen's with one particular file, with another file i dont have this issue.
With Slic3r 1.3.0-dev i don't have this issue.

#### STL/Config (.ZIP) where problem occurs
[slic3r_issue.zip](https://github.com/prusa3d/Slic3r/files/674164/slic3r_issue.zip)

bubnikv, 2017-01-02:
Hmm, I cannot reproduce it on Windows 64bit. I cannot reproduce the issue on Arch Linux with our official Linux binary build
https://github.com/prusa3d/Slic3r/releases/download/version_1.31.6/Slic3r-1.31.6-prusa3d-linux64-full-201611202210.tar.bz2

Did you compile Slic3r yourself? Does our official binary work for you?

myckeul, 2017-01-02:
Hi,

This version cause a segfault on my system, but the previous version
with your binary build
"Slic3r-1.31.2-prusa3d-linux64-full-201611011501.tar.bz2" work properly
but i dont have the gui, only cli work correctly.

Regards,

bubnikv, 2017-01-02:
> This version cause a segfault on my system, but the previous version
with your binary build

Would you please share any details?

Where did you get the Slic3r binary with which you initially encountered the problem?

> "Slic3r-1.31.2-prusa3d-linux64-full-201611011501.tar.bz2" work properly
but i dont have the gui, only cli work correctly.

You will find out more, if you force slic3r to run in the GUI mode with a --gui parameter.

myckeul, 2017-01-02:
> Would you please share any details?

Yes : 

`./slic3r : ligne 4 :  7083 Erreur de segmentation  (core dumped) "$DIR/bin/perl5.22.0" "$DIR/bin/slic3r.pl" $@`

But i have find the reason of this segfault, I'm French, and the directory of my download folder is "Téléchargements" ... 

If i move the directory of Slic3r in another directory without accented chars Slic3r work correctly.

The issue with "processing triangulated mesh" don't happens with the binary version. 

Regards

VBen, 2017-01-04:
I have the same problem on some stl models.
I'm using linux and a self compiled version (sourcecode 03.01.2017)
The current release version works well
[config_stls.tar.gz](https://github.com/prusa3d/Slic3r/files/685187/config_stls.tar.gz)

bubnikv, 2017-02-02:
This seems to be a problem in the detection of extra perimeters.
Fixed by @alexr alexrj@8f0d8f2
See also alexrj#3613 alexrj#3614

closed 2017-02-02 by bubnikv

JopperTom, 2020-03-31:
I am having this issue with PrusaSlicer-2.2.0 installed from download. Spinner with (Not Responding) in Windows 10 after selecting Slice.
http://www.thingiverse.com/thing:2064359
[image]

bubnikv, 2020-03-31:
Please provide a 3MF. The following slices nicely with PrusaSlicer 2.2.0.

[new-pangolin_body-all.zip](https://github.com/prusa3d/PrusaSlicer/files/4407086/new-pangolin_body-all.zip)

JopperTom, 2020-04-02:
> I am having this issue with PrusaSlicer-2.2.0 installed from download. Spinner with (Not Responding) in Windows 10 after selecting Slice.
> http://www.thingiverse.com/thing:2064359
> [image]

> Please provide a 3MF. The following slices nicely with PrusaSlicer 2.2.0.
> 
> [new-pangolin_body-all.zip](https://github.com/prusa3d/PrusaSlicer/files/4407086/new-pangolin_body-all.zip)

I restarted my system and PS, and was unable to reproduce this again. I will let you know if it happens again.

Thanks.