prusa3d/PrusaSlicer #76: version `CXXABI_1.3.8' not found

misan, 2016-12-27:
### Version
Prusa Slic3r 1.3.8_

### Operating system type + version
Ubuntu 14.04 64bit

### Behavior
Downloaded the version tar, and I got this message when trying to run Slic3r

Can't load './prusa-Slic3r/lib/site_perl/5.22.0/auto/Slic3r/XS/XS.so' for module Slic3r::XS: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by ./prusa-Slic3r/lib/site_perl/5.22.0/auto/Slic3r/XS/XS.so) at ./prusa-Slic3r/lib/site_perl/5.22.0/XSLoader.pm line 70.
 at /home/misan/Documentos/prusa-Slic3r/lib/site_perl/5.22.0/Slic3r/XS.pm line 22.
Compilation failed in require at ./prusa-Slic3r/lib/site_perl/5.22.0/Slic3r.pm line 54.
BEGIN failed--compilation aborted at ./prusa-Slic3r/lib/site_perl/5.22.0/Slic3r.pm line 54.
Compilation failed in require at ./bin/slic3r.pl line 15.
BEGIN failed--compilation aborted at ./bin/slic3r.pl line 15.

misan, 2016-12-27:
That fixed it:

sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt-get update
sudo apt-get install gcc-4.9 g++-4.9

closed 2016-12-27 by misan

AvdN, 2018-12-22:
> sudo add-apt-repository ppa:ubuntu-toolchain-r/test
> sudo apt-get update
> sudo apt-get install gcc-4.9 g++-4.9

This solve my issue on Mint 17, but it should be noted that the "normal" Slic3r appimage () doesn't have this problem (Slic3r-1.3.0-x86_64.AppImage)