prusa3d/PrusaSlicer #77: Slic3r Configuration Assistant

probonopd, 2016-12-28:
### Version
1.31.6-prusa3d

### Operating system type + version
ubuntu-16.04-desktop-amd64.iso

### Behavior
When starting Slic3r for the very first time, a Configuration Assistant starts up and asks questions that the application should know based on the machine the user has. For example:
* __G-code flavor__ (RepRap (Marlin/Sprinter/Repetier))
* __Bed Size__ (Shape, Size x, Size y, Origin x, Origin y)
* __Nozzle Diameter__ (mm)
* __Filament Diameter__ (mm)
* __Temperature__ (degrees celsius)
* __Bed temperature__ (degrees celsius)

While these settings may be welcome for advanced users experimenting, it would be nice if there was a way to just select a __machine__ such as "Original Prusa i3 MK2 3D Printer" and a filament type such as __Original Prusa PLA__ and be done with it.

It would greatly add to the out-of-the-box experience, a.k.a. "polish", if default settings for mainstream machines like the Original Prusa i3 MK2 3D Printer would be working without having to delve into technical details.

Compare this with an inkjet printer driver that would ask you to enter the print surface in centimeters, the type of print head, and the chemical structure of the OEM ink before you can start to print...

_Is this a new feature request?_ Yes

bubnikv, 2017-01-02:
We have a drivers package for windows, which installs the config files and pre-configures the default printer and filament for you. We don't have that for Linux or MacOS as of today.

You are probably aware, that for the Prusa Research printers you will find the Slic3r configs here.
https://github.com/prusa3d/Slic3r-settings

I believe if you unpack them into ~/.slic3r , you will have your Slic3r configured the same way it works on Windows when you install the drivers package, but I did not test that yet.

How did you install Slic3r on your ubuntu?

bubnikv, 2017-12-10:
This request has been implemented with https://github.com/prusa3d/Slic3r/commit/657f2734f19da1d6483b9d70b5d8102e727bc34b
https://github.com/prusa3d/Slic3r/commit/98fdb46da9741a4624a427b00ef355b7ee25abc6
Closing this issue, leaving https://github.com/prusa3d/Slic3r/issues/425 open.

closed 2017-12-10 by bubnikv