prusa3d/PrusaSlicer #79: Number of solid top/bottom layers ignored

NikoDelarich, 2016-12-29:
### Version
1.31.6-prusa3d-win64

### Operating system type + version
Windows 10 Pro 64-bit, all updates

### Behavior

I'm trying to print a model that consists of two separate islands that are joined after ~1mm.
The number of solid top & bottom layers is ignored for the inner/smaller island: A single solid top & bottom layer is printed even though the top layers are set to 12 & the bottom layers are set to 10.
For the outer/larger island this works correctly.

* Load attached STL
* Slice at 0.10mm DETAIL
* Compare top & bottom layers of inner & outer structure
* _Expected Results_: Same number of top & bottom layers in both structures
* _Actual Results_: Inner structure has only a single top & bottom layer

_Is this a new feature request?_ No

Second layer (inner structure is left, outer structure is right):
[image]

Second to last layer:
[image]

#### STL/Config (.ZIP) where problem occurs
[testcircle.zip](https://github.com/prusa3d/Slic3r/files/676994/testcircle.zip)

bubnikv, 2017-01-02:
Thanks for a great test case. This is a duplicate of https://github.com/prusa3d/Slic3r/issues/60 .

It is a critical bug and I shall release a patch for it.

bubnikv, 2017-01-02:
Fixed by https://github.com/prusa3d/Slic3r/commit/17db5bff8d722369b648d41644782772f285ecd0

closed 2017-01-02 by bubnikv