prusa3d/PrusaSlicer #80: Wrong (original) Icon shown when (Ubuntu) package slic3r is installed

SparkyCola, 2017-01-02:
### Version
1.31.6, from released binary
### Operating system type + version
Ubuntu 16.10 64-bit

### Behavior
The wrong (original) Slic3r Icon is shown when launched via  the "slic3r" launch script in the folder, but the Prusa Version is definitely started (Version Name in lower status var of the window and about dialog, also the new options announced [here](http://www.prusaprinters.org/introducing-slic3r-prusa-edition/) are there.
The wrong icon occurs only when the slic3r package is installed, after purging slic3r it didn't occur anymore.

bubnikv, 2017-03-23:
Is it still an issue? Does it happen with the appimage?

bubnikv, 2017-03-23:
I mean, does it happen with the 1.33.8 appimage?

SparkyCola, 2017-03-24:
No it works with the image.

closed 2017-03-24 by SparkyCola