prusa3d/PrusaSlicer #81: "With sheath around the" option unable to be deselected with mouse

nebbian, 2017-01-03:
### Version
1.31.6-prusa3d

### Operating system type + version
Mac OS 10.10.5 (Yosemite)

### Behavior
I am unable to select or deselect the "With sheath around the" checkbox using the mouse.

I am, however, able to select and unselect it using the keyboard.

This is not a new feature request.
[image]

bubnikv, 2017-03-30:
I have updated the wxWidgets library to 3.0.3-beta on the build server. Please let me know whether the attached alpha build solves your issue.

[Slic3r-1.33.8.153-prusa3d-full-g36416d7-201703300827.zip](https://github.com/prusa3d/Slic3r/files/882151/Slic3r-1.33.8.153-prusa3d-full-g36416d7-201703300827.zip)

nebbian, 2017-03-30:
Thanks for this, however it appears as if that zip is corrupted.  I get the following error when decompressing from the command line:
Archive:  /Users/ben/Desktop/Slic3r-153-prusa3d-full-g36416d7-201703300827.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.

bubnikv, 2017-03-30:
Would you please try the currently released 1.34.0 alpha? THanks.

nebbian, 2017-03-31:
Thanks @bubnikv however the problem is still present.  I've made a screen recording that you can view here:
https://www.youtube.com/watch?v=a4hdnhahkoo&feature=youtu.be

bubnikv, 2017-06-08:
This is an interesting bug, because the "XY separation" label is split to two lines as well and it is displayed correctly even on your computer. I am not sure what I can do. I think I will check on other OSX boxes around.

[image]

jhitesma, 2017-06-08:
Hadn't seen this issue before, but I've experienced it intermittently on macOS Sierra with versions up to 1.34.1.24

Except it's not limited to just the Sheath around the support, it's every checkbox on the support materials page sometimes stop working and I can't check or uncheck any of them.  Restarting slic3r usually clears it up and they start working again.  

I haven't been able to figure out what triggers it since it doesn't happen to me every time (and I don't use support very often so I'm not normally on that page.)

bardiir, 2017-06-09:
Have seen this too, switching the tab to something different than support for this example and then switching back does solve it for me every time however.