prusa3d/PrusaSlicer #83: missing a layer ?

Richie76, 2017-01-06:
### Version
1.31.6-prusa3d-win64

### Operating system type + version
win 10 64bit Pro - full up to date

### Behavior
I´m printing a lot of 3DBenchys with different material.
It seems like the printer is skipping a layer and I´m out of my knowledge. (I´ve asked for help on the prusa forums too - maybe it´s the printer).
see first picture:
[http://www.loopback.at/wp-content/uploads/2017/01/DSC_0006-6.jpg](url)

left and top are showing the "error" and are the same PLA material in a different colours (obvious ;-))
bottom one is a different brand PLA and the right one is ABS - those don´t show this error.

Print setting are the same on all 4 prints - hence no cooling for the ABS one and 3 layers without fan on left and top (those with the errors).

next picture:
[http://www.loopback.at/wp-content/uploads/2017/01/DSC_0008-4.jpg](url)
another test with an extrusion multiplier of 0.9 (others were at 1.0) - model looks a bit better overall but still this missing layer.

The layer which is "missing" (all benchys have the same dimensions - so a missed line should be noticeable) is areoung the high of the "deck" -> see picture:
[http://www.loopback.at/wp-content/uploads/2017/01/benchy_error.jpg](url)

in slic3r everything looks good and also the generated gcode shows no error - either on gcode.ws nor in octoprint. I can test without octoprint tomorrow ... but this gave me no issues since now.

Other models with this particular PLA are quite nice and error free.

Is this a slic3r issue ? (is it maybe the 3 layers no cooling option ?)

#### STL/Config (.ZIP) where problem occurs
the zip contains the stl, the gcode and the config.ini
[config.zip](https://github.com/prusa3d/Slic3r/files/688707/config.zip)

Richie76, 2017-01-06:
Quick follow up:
I did another test with a different PLA (no-name branded) and it came out really good.
Then i did another test with another color of the "faulty material" - still with no cooling at the first 3 layers and 0.8 extrusion multi. Same error.
I´ll try another one with cooling from 2nd layer on. Can´t think that this makes a difference.
Also the gcode looks ok.
Last test will be to use another material setting with this material. Makes a total of 4 benchys for today ;-(

bubnikv, 2017-01-06:
Sorry, I cannot access the images, they got lost. I believe you can drag and drop the images directly into the response edit window.

Everybody loves the benchies. I love them dearly, I am happy for every benchy printed!

Richie76, 2017-01-06:
first post edited to show the pictures.

I do love the benchys - but this issue is quite annoying.

Sebastianv650, 2017-01-06:
I don't think this is related to the slicer - especialy when the same gcode runs without problems with other filaments.
I have seen this also on my printer, it's always due to bad Firmware experiments (my fault) or tangled filament or simmilar filament feed issues. Try to unroll some meter of filament from the "bad" spools and see if it blocks somewhere due to a "tangled" section. Can happen if you don't secure the free end of the spool when not used..

bubnikv, 2017-01-06:
I don't think this is a Slic3r issue. It may be a firmware or a hardware issue, it may be a bad luck with the component tolerances of the printer parts. Namely, it seems to me, that for some reason the Z axis skipped steps. We experienced issues with the Z stepper motor cables being damaged by squeezing them too tight against the threaded rods.

Did you use the "standard" Prusa i3 MK2 profiles for the Slic3r, or did you modify them?
Do you use a stock MK2 firmware or did you tinker with it? Do you use the "silent" or the "power" mode of the printer? What firmware version?

Richie76, 2017-01-06:
Hi,
the printer is stock except the coldend-fan (have replaced it with some 3015 type - same or better cooling with a lower pitch noise). Firmware is the latest.

modified slicer profiles but the same profile on all prints.

I´ve measured some benchys and they are all the same size (give or take a few tenth mm) - so its not a skipped layer I believe. The are before this strange gap happens is narrower than on the correct benchys. Currently I´m printing one at lower table temp and higher nozzle temp and it doesnt show the error.

It seems that the 65°C i had on the bed causes the outer walls to lean in maybe due to to much cooling (the gap happens in the first layer of the "deck" - so the printer takes a lot of time at the back of the benchy - causing the front walls to lean in.

I´m sorry to have bothered you - I´ll look into the temps - but for now its "solved" I guess.

Thanks for all the suggestions which kinda lead me to solving the issue.

And i´ll give you 2 pictures of my benchy collection ;-) 
[image]
[image]

Thanks for your time and help,
Richard

bubnikv, 2017-01-06:
> The are before this strange gap happens is narrower than on the correct benchys. Currently I´m printing one at lower table temp and higher nozzle temp and it doesnt show the error.

Some filaments require a higher nozzle temperature than others. PLA is quite sensitive, if you are too cold, the viscosity of PLA gets too high and the extruder wheel may start grinding the filament. This often happens after repeated retractions. So this effect is sensitive to the print geometry due to the retractions.

> It seems that the 65°C i had on the bed causes the outer walls to lean in maybe due to to much cooling (the gap happens in the first layer of the "deck" - so the printer takes a lot of time at the back of the benchy - causing the front walls to lean in.

You may get similar effects if you just open the door or window.

> I´m sorry to have bothered you - I´ll look into the temps - but for now its "solved" I guess.

No problem. Good luck with your MK2.

closed 2017-01-06 by bubnikv