prusa3d/PrusaSlicer #86: Wrongly sliced part

konvoj, 2017-01-09:
### Version
Slic3R Prusa Edition 1.31.6

### Operating system type + version
Windows 8.1

### Behavior
Some parts are sliced wrongly. when trying to print part:
http://www.thingiverse.com/thing:1933258
according to the instructions (1 layer thick, no bottom, no top layer)

The result of slicing with Slic3r gives some extra material on some spots making "columns" raising from the part. Please see the photo with comparison of a part sliced with Slic3r and Simplify3D. This is not a printer issue, nor a material issue; I tried on several similar 3D models and printing from PLA, ABS, PETG, always with the same result. I checked the visualization in slicer to see if these columns would be noticeable in the preview, but they are not, I haven't gone through the G-code though.
[image]

Not a new feature.

#### STL/Config (.ZIP) where problem occurs
STL file link above.

bubnikv, 2017-01-09:
I doubt this is a Slic3r settings. I am getting such "columns" from a leaky extruder, or if the retraction settings is not optimal. What printer are you using? What settings? I bet it is not the Prusa I3 MK2.

konvoj, 2017-01-09:
Original Prusa I3 MK2, retraction 0.8 mm, which is default I think. The same retraction settings is used in Simplify3D and print doesn't have these issues.

Sebastianv650, 2017-01-09:
This is a classical oozing problem. If it happens this way or not depends in the travel path. This tiny "strings" can only build up if the travel path will always move over the same spot.
But before somebody recommends to change the travel path system, don't do that. If you look carefuly to the Spimplify print, you should be able to see tiny zits or something where the travel move hits the new perimeter. The oozing material will always be somwhere, the question is only: where.

Improving the retraction parameters will help.

bubnikv, 2017-01-10:
> Original Prusa I3 MK2, retraction 0.8 mm, which is default I think. The same retraction settings is used in Simplify3D and print doesn't have these issues.

I believe you have somehow messed up the Slic3r parameters. Are you using the parameters provided by Prusa3D? 

https://github.com/prusa3d/Slic3r-settings

Our print expert Jindra ran this particular print yesterday. Due to the power outage only the first 1.5cm got printed, but we don't see the effect you reported.

bubnikv, 2017-01-10:
We cannot reproduce your issue. I think you have your Slic3r profiles messed up. Closing.

[image]

closed 2017-01-10 by bubnikv