prusa3d/PrusaSlicer #87: Filenames generated by Slic3r when directly sent to Octoprint

konvoj, 2017-01-09:
## Version
Slic3R Prusa Edition 1.31.6

### Operating system type + version
Windows 8.1

### Behavior
When sending a sliced parts through Octoprint to the printer, the file generated has always the name of the first stl file imported into Slic3r. When printing bed full of different components, this could be quite misleading. Sometimes when trying different settings or materials at once, one needs to generate G-code to the local drive, rename to e.g. PLA-my-print-220-50 or ABS-my-print-255-100 and then upload to Octoprint manually. This is time consuming mainly when testing more settings.
Proposed solution is to add a simple text insert box to be able to change the filename manually before sending to Octoprint.
Another proposal would be that the generated filenames always start with material abbreviation so the different files are easily recognizable in Octoprint list. 
Yet another proposal is that the filenames shall end with the hotend/bed temperature setting.
All of these choices shall have a tickbox in settings so they can be turned individually on/off.

New feature.

#### STL/Config (.ZIP) w
N/A

Richie76, 2017-01-19:
Hi there,

Does this information help ?
[https://github.com/alexrj/Slic3r/wiki/FAQ#how-can-i-specify-a-custom-filename-format-for-output-g-code-files](url)

I´ve set mine to _[input_filename_base]_[filament_preset].gcode_ which shows still the first added stl but also the filament name.

Cheers, Richard

closed 2018-06-21 by vojtechkral

bubnikv, 2018-06-21:
It will be part of 1.40.1.