prusa3d/PrusaSlicer #88: Add more details about missing Boost library during build

eyal0, 2017-01-09:
### Version
version_1.31.6-78-gd9ea3df

### Operating system type + version
Raspbery Pi Debian 8

### Behavior
When building from sources, the error message is not useful: 

Configuring /home/pi/Slic3r/xs ... Slic3r requires the Boost libraries. Please make sure they are installed.

If they are installed, this script should be able to locate them in several
standard locations. If this is not the case, you might want to supply their 
path through the BOOST_DIR environment variable:

    BOOST_DIR=/path/to/boost perl Build.PL

Or you may specify BOOST_INCLUDEPATH and BOOST_LIBRARYPATH separatly, which
is handy, if you have built Boost libraries with mutliple settings.

It's unclear which libraries need to be installed.  The libraries needed by the alexrj instructions are not sufficient.  You should mention:

 sudo apt-get install libboost-thread-dev libboost-system-dev libboost-filesystem-dev libboost-log-dev

It's the last one, log, that is needed by prusa3d build but not alexrj so the alexrj instructions won't work for this fork.

dmopalmer, 2017-02-20:
Also, I think that BOOST_LIBRARYPATH in the error message should be BOOST_LIBRARYDIR

bubnikv, 2017-03-23:
Thanks, I have updated Build.PL with the list of boost libraries required,
I have also fixed the BOOST_LIBRARYPATH to BOOST_LIBRARYDIR.

closed 2017-03-23 by bubnikv