prusa3d/PrusaSlicer #89: Better anchoring of vertical towers into the top surfaces

bubnikv, 2017-01-10:
Vertical towers placed over a flat surface may not be anchored well enough into an object if printed with a plastic, that shows less than ideal layer to layer bonding. For example, following part of the I3 MK2 printer is quite fragile if printed with ABS-T.

[image]

[y-motor.zip](https://github.com/prusa3d/Slic3r/files/696742/y-motor.zip)

[image]

As seen on the following two images, the tower is placed flat over the top infill.
[image]
[image]

This is a good enough strategy for PLA, but not a good idea for ABS. For ABS, it would be better to extend the perimeters of the tower one or two layers into the top surface of the base, so the threads of the tower overlap one over the other and the infill as well. Yet better, it would be great if the infill could continue inside the sunk tower and continue into the infill of the base, so the infill lines overlap exactly.

The proposed strategy shall be optional as the old strategy would probably achieve nicer top surface of the base and faster prints.

phantomgnom, 2020-02-19:
Hi Bubnikv, it is also no good strategy for PLA. Everything breaks. Have you seen, that this only happens for smaller objects < 7mm in diameter. > 7mm gets anchored. So in general, it works but not for small objects. Other slicers starting anchoring at 1 mm diameter.

Xaymar, 2021-05-07:
I ran into this issue with PrusaSlicer 2.3.1, no matter what material I used. The parts do not "stick" together, and create what appears to be an intentional weak point. Oddly enough, it doesn't always do it, something it does decide to leave a tiny gap:
[image]

To me this apparently intentional weakening of parts makes no sense, I have to turn all my parts on their side to get decent GCode from PrusaSlicer. Or alternatively use materials that can be heated to form better bonds later on.

Sineos, 2021-05-08:
I run into this problem also. In such cases I will have to resort to Cura. Among my "go to" slicers Cura has the best strategy IMO, although I still find it less then optimal.

Following images show the "top layer" followed by the first layer of the tower:

**Prusa Slicer:**
[image]

**Cura**
[image]

**Simplify3D**
[image]

In my opinion all perimeters of the tower should start on the same same layer as the top layer.

closed 2026-09-29 by prusaslicer-bot[bot]