prusa3d/PrusaSlicer #90: Latest master segfaults with perl 5.24.1

alexxy, 2017-01-10:
### Version
slic3r.pl --version 
1.31.6
$ git describe --tags 
version_1.31.6-78-gd9ea3df8

### Operating system type + version
Linux

### Behavior
$ LANG=C slic3r.pl --gui 
Platter - update
Segmentation fault

However if i start it with no-plater option it works

alexxy, 2017-01-10:
And seems like it crashes because of perl gl 0.70 package

CReimer, 2017-01-19:
This problem can be fixed with this pull request: https://github.com/alexrj/Slic3r/pull/3575
I hope this gets merge here to before the next stable release

closed 2017-01-20 by bubnikv

bubnikv, 2017-01-20:
Please let me know whether it fixes your crash. I did not test it against an OpenGL 0.7 Perl module as I am using a modified older OpenGL module on all platforms to support multi sample anti aliasing.