prusa3d/PrusaSlicer #91: Support generation takes too long, sometimes never finishes

bubnikv, 2017-01-11:
Slic3r Prusa Edition 1.31.6

Please see https://github.com/alexrj/Slic3r/issues/3656 for a set of test objects, for which the current support generator does not produce supports in a reasonable time.

alexxy, 2017-01-11:
Does new support code already awailable in master? I just build this version, but it seems support doesnt get generated at all
[image]
[image]
[image]

bubnikv, 2017-01-11:
Yes, it is there. The code is not properly tested yet, maybe it does not
work with some parameter sets.
I am using an overhang threshold around 33 to 45 degrees, gap between the
print and the object around 0.1mm.

alexxy, 2017-01-11:
Ok. Can you share parameters for support? I tryed many combinations but result is always as on screenshot

bubnikv, 2017-01-11:
[config.zip](https://github.com/prusa3d/Slic3r/files/699376/config.zip)

My college found a bug in the 1st layer thickness, so for some models the current code does not honour the 1st layer thickness. Also there is a bug in that the brim crosses the 1st support layer.

bubnikv, 2017-01-20:
@alexxy You should be able to compile the master and print supports. The current code base seems to be stable in regard to the single extruder prints. I am currently printing your 2l15 model at 150% of its size. I am using the following parameters.

[image]

I tried to print the 2l15 model at a 100% size with some older version, which had issues with the tolerances of the top / bottom layers. I was able to remove the supports from the outside, but the supports inside the ball were difficult. The current master shall be all right in regard to the contact layer tolerances and printing it a bit bigger should make it easier to remove the supports from the center of the structure. I am really curious how it comes out :-)

bubnikv, 2017-02-09:
Hopefully fixed in the current master. Will release a beta tomorrow.

closed 2017-02-09 by bubnikv