prusa3d/PrusaSlicer #92: Create gyroid support structure

mrjohnc, 2017-01-11:
Please create gyroid infill that is the shape as graphene. This shape has been shown to be very strong as a 3D printed object and if used as an infill would presumably create very strong 3D prints 

http://itsybitsymag.com/2017/01/11/mit-research-ten-times-strong-lightweight-graphene/

bubnikv, 2017-01-12:
The structure has been described in a year 2013 in 
Int J Adv Manuf Technol (2013) 66:1247 – 1254
A new approach to the design and optimisation of supportstructures in additive manufacturing
G. Strano & L. Hao & R. M. Everson & K. E. Evans

Funny thing, I cannot find the paper published by the MIT folks, I wonder whether they referenced the paper I mentioned.

mrjohnc, 2017-01-12:
Here it is :)

http://advances.sciencemag.org/content/3/1/e1601459.full

mrjohnc, 2017-01-13:
I printed a gyroid shape where all surfaces are 0.4mm thick (the size of my nozzle), I think this is the same as what the infill would be like, the model is 40mm x 40mm.

It is very very strong indeed, I think I could maybe stand on it and it would be ok, it weighs very very little indeed

I got the model from here http://www.3dprintmath.com/chapters/6-Surfaces

[image]

landodragon141, 2017-01-13:
That's really cool!

mrjohnc, 2017-01-13:
I also submitted this request on the main Slic3r issues page, I'm not sure if there is a way to share issues between repositories?

https://github.com/alexrj/Slic3r/issues/3666

bubnikv, 2017-01-13:
> alexrj#3666 <https://github.com/alexrj/Slic3r/issues/3666>

There is no way to share issues between repositories IMHO, but github
automatically references one issue gainst the other, which is cool.

lordofhyphens, 2017-01-15:
Yeah, which ever repo gets it first, it can get into the other one without too much fuss.

mrjohnc, 2017-03-09:
I created some test cubes to compare the strengths of infill, any ideas how to compare strengths? 

http://www.thingiverse.com/thing:2091011

landodragon141, 2017-03-09:
You need some way to exert a controlled force on the cube. Do you have access to weights of some kind? You could put a bucket on it and gradually fill it with water. That's about the best poor mans idea I can think of. Otherwise you could try and visit the engineering department of a local college. They would very likely have something that could perform a test.

werewolf1479, 2017-03-09:
landodragon is correct.  You want to be controlled (have you calculated the infill % of your gyroid fill?) and as incremental as possible.  I'd make a little plywood jig from two 1/2" or thicker boards.  One would be the exact size of your cube or a touch smaller, and that would be centered on a larger board that you can start stacking weight on.  

A slightly more complicated method would be to get a decent industrial scale, mount a scissor jack (from your car) upside down from a constructed wood gantry.  Put the cube on the scale, tare it or note the weight, then start crushing it with the scissor jack.  

A similar but much more expensive option would be to set something up like this:  https://www.youtube.com/watch?v=ZBmxkWK_OFA

The engineering dept idea is good, the mech e's may have a hydraulic press.  You may get roped into doing a demo for an interested professor as well :)

supermerill, 2018-01-08:
Mostly done
[image]

code here:
https://github.com/supermerill/Slic3r/tree/Gyroid

lordofhyphens, 2018-01-08:
I look forward to seeing the pull request against alexrj/Slic3r :)

mrjohnc, 2018-01-08:
@supermerill amazing, so if I increase the infill percentage the pattern inside gets smaller?

supermerill, 2018-01-08:
https://imgur.com/a/fwwTe
[image]

i will do a pull request also in the alexjr branch if i don't fail to install the compiling environment.

@mrjohnc  yes
10%:
[image]

30%:
[image]

mrjohnc, 2018-01-09:
@supermerill looks great :) fingers crossed, let me know if I can be of any help

lordofhyphens, 2018-01-09:
Shouldn't be too bad. Download Slic3r-perl from wiki and put everything in
path.

I know that there is a setup for building on Windows with vc++

supermerill, 2018-01-09:
current pull test:
(160*20*20, 1 perimeter, 2 bot/top layers, nozzle:0.4mm, lh: 0.2mm)
cubic 9.67g 41min 14kg
gyroid 9.81g 42min 13kg
[image]

i will try with some other parameters tomorrow.

mrjohnc, 2018-01-09:
There are lots of test you could try, Thomas Sanladerer runs a series on Youtube testing different filaments that might be useful. It would be super cool for him to do similar tests with different infill patterns. 

Here's a video he did on explaining different tests he does. 

https://www.youtube.com/watch?v=nlIFMy4vIeM

supermerill, 2018-01-28:
I created an experimental build here https://www.reddit.com/r/3Dprinting/comments/7tkkd9/experimental_need_some_beta_tester_for_slic3r/
to let people test before i pull request. 
I don't know what kind of gyroid thickness to submit.

bubnikv, 2018-02-21:
> current pull test: ...

It seems to me that the biggest benefit of the gyroid infill is its rigidity in all axes (isotropy), not strength.

bubnikv, 2018-03-14:
Implemented by @supermerill, merged into the current master. It will be part of the upcoming Slic3r 1.34 release.

closed 2018-03-14 by bubnikv

tkircher, 2018-05-16:
The referenced paper above should be:

http://advances.sciencemag.org/content/3/1/e1601536

bubnikv, 2018-05-16:
I suppose you wanted us to fix the release page? I did that. Thanks.

ges110, 2019-07-27:
It's great to see the Gyroid used in this way. 
There was actually work about a decade ago that showed that the strength of the Gyroid comes through the homogeneity of its structure:
https://www.sciencedirect.com/science/article/pii/S0142961211006776
What's interesting is that the gyroid is quite strong in two ways: either when you build it as a warped sheet (what's described above) but also when you build it as a network-like structure.
Maybe there's some food for thought!
Also there are plenty of other minimal surfaces that are similar in their structure, see paper above or earlier work by others.
Cheers

bubnikv, 2019-07-31:
> What's interesting is that the gyroid is quite strong in two ways: either
when you build it as a warped sheet (what's described above) but also when
you build it as a network-like structure.

Thanks for heads up. A dual structure to gyroid may be useful for the SLA
infills.