prusa3d/PrusaSlicer #93: Tests fail when compiling on Raspberry Pi

eyal0, 2017-01-11:
### Version
version_1.31.6
### Operating system type + version
Raspberry Pi Debian 8

### Behavior
Built Slic3r according to instructions on desktop linux and on Raspberry Pi.  Build is successful in both cases.  All tests pass on Linux build but some tests fail on Raspberry Pi.

eyal0, 2017-01-11:
I've been able to track the problem down to the sort line in FillRectilinear2.cpp, line 991.  I'm not sure if it's because the RPi uses 4 byte values where the desktop is using 8 byte or if it's because of the different versions of c++ but the result of the sort is different on each processor.  This is because the less than operator "<" is not consistent.

This is output from the desktop:

```
(gdb) x/20x &sil.intersections[0]
0x7e4110:       0x00000000      0x00000000      0x00000001      0x00000000
0x7e4120:       0xc4ce4ff0      0xfffffea0      0x02fa0482      0xffffffff
0x7e4130:       0x00000000      0x00000000      0x00000000      0x00000000
0x7e4140:       0x00000003      0x00000000      0xeed1fdeb      0xfff72261
0x7e4150:       0x02fa0482      0xffffffff      0x00000000      0x00000000
(gdb) p/x sil.intersections[0]
$31 = (__gnu_cxx::__alloc_traits<std::allocator<Slic3r::SegmentIntersection> >::value_type &) @0x7e4110: {iContour = 0x0, iSegment = 0x1, pos_p = 0xfffffea0c4ce4ff0, pos_q = 0x2fa0482, 
  type = 0xffffffff, consumed_vertical_up = 0x0, consumed_perimeter_right = 0x0}
(gdb) p sizeof(sil.intersections)
$32 = 24
(gdb) x/20x &sil.intersections[2]
0x7e4160:       0x00000001      0x00000000      0x00000001      0x00000000
0x7e4170:       0xd47a8082      0xfffff26d      0x02f1b882      0xffffffff
0x7e4180:       0x00000000      0x00000000      0x00000001      0x00000000
0x7e4190:       0x00000003      0x00000000      0xe2bb5e7e      0xfff7474e
0x7e41a0:       0x02f1b882      0xffffffff      0x00000000      0x00000000
(gdb) p/x sil.intersections[2]
$33 = (__gnu_cxx::__alloc_traits<std::allocator<Slic3r::SegmentIntersection> >::value_type &) @0x7e4160: {iContour = 0x1, iSegment = 0x1, pos_p = 0xfffff26dd47a8082, pos_q = 0x2f1b882, 
  type = 0xffffffff, consumed_vertical_up = 0x0, consumed_perimeter_right = 0x0}
(gdb) p sil.intersections[0] < sil.intersections[2]
$34 = false

```
This is on RPi:

```
(gdb) x/20 &sil.intersections[0]
0x1d4d50:       0x00000000      0x00000001      0xc4ce4ff0      0xfffffea0
0x1d4d60:       0x02fa0482      0xffffffff      0x76ff0000      0x76fff958
0x1d4d70:       0x00000000      0x00000003      0xeed1fdeb      0xfff72261
0x1d4d80:       0x02fa0482      0xffffffff      0x76ff0000      0x76fff958
0x1d4d90:       0x00000001      0x00000001      0xd47a8082      0xfffff26d
(gdb) p/x sil.intersections[0]
$57 = (__gnu_cxx::__alloc_traits<std::allocator<Slic3r::SegmentIntersection> >::value_type &) @0x1d4d50: {iContour = 0x0, iSegment = 0x1, pos_p = 0xfffffea0c4ce4ff0, pos_q = 0x2fa0482, type = 0xffffffff, 
  consumed_vertical_up = 0x0, consumed_perimeter_right = 0x0}
(gdb) p sizeof(sil.intersections)
$58 = 12
(gdb) x/20x &sil.intersections[2]
0x1d4d90:       0x00000001      0x00000001      0xd47a8082      0xfffff26d
0x1d4da0:       0x02f1b882      0xffffffff      0x76ff0000      0x76fff958
0x1d4db0:       0x00000001      0x00000003      0xe2bb5e7e      0xfff7474e
0x1d4dc0:       0x02f1b882      0xffffffff      0x76ff0000      0x76fff958
0x1d4dd0:       0x00000000      0x00000089      0x00000000      0x00000001
(gdb) p/x sil.intersections[2]
$59 = (__gnu_cxx::__alloc_traits<std::allocator<Slic3r::SegmentIntersection> >::value_type &) @0x1d4d90: {iContour = 0x1, iSegment = 0x1, pos_p = 0xfffff26dd47a8082, pos_q = 0x2f1b882, type = 0xffffffff, 
  consumed_vertical_up = 0x0, consumed_perimeter_right = 0x0}
(gdb) p sil.intersections[0] < sil.intersections[2]
$60 = true

```
Even though the of each intersection are the same, the comparison returns a different result.

Maybe the less than function isn't defined for the intersection type explicitly so it's using some implicit definition and that definition isn't consistent across platforms?

In the same area of the code, I see a commented-out section of code to perform a bubble sort.  Maybe this bug was encountered before and there was an attempt to fix it?

I think than < needs to be overloaded correctly so that sorting will work.

eyal0, 2017-01-11:
Oops, I see that I was on an old version.  I also found the operator< definition in FillRectilinear2.cpp.  I'll build again on version 1.33.2 both at desktop and RPi and see if the problem goes away.

eyal0, 2017-01-12:
Upgrading to 1.33.2 did not solve the problem.  I'll investigate the compare procedure.

bubnikv, 2017-01-12:
The SegmentIntersection == and < operators evaluate the rational
expressions.
If two numbers a and b are expressed as ratios of a1/a2 and b1/b2, instead
of comparing
a1/a2 < b1/b2 one compares a1 * b2 < b1 * a2

The routine has been tested on 32bin mingw on Windows, so I wonder what
could cause this not work on 32bit Raspberry. It is quite difficult for me
to find out without actually debugging the code on the Raspberry.

eyal0, 2017-01-12:
I've sent a pull request.  The problem is that char was assumed to be signed.

The code that I used to test this change is here: https://github.com/eyal0/Slic3r/tree/fix_rpi_less_than_testing

I made a small test harness to test just this bug.  That code is here:
https://github.com/eyal0/Slic3r/tree/fix_rpi_less_than_testing

The file "TestFillRectilinear2.cpp" has instructions on how to compile and run a test that will succeed on my desktop but fail on RPi.  After the patch, both RPi and desktop pass just that test.  The test is copied from fill.t .

I'm rebuilding and running the entire test suite now to see if all the other tests pass.  (It takes ~1 hour to compile Slic3r on RPi.)  I'll let you know the results.

bubnikv, 2017-01-12:
Maybe my urge to use a "char" instead of "int" was caused by my parallel involvement with the Marlin 8bit firmware :-)

closed 2017-01-12 by bubnikv

eyal0, 2017-01-12:
Great news, all tests now pass on RPi.

bubnikv, 2017-01-12:
Thanks to you, great job.

May I ask you why do you want to run Slic3r on such a low computing power device? Do you want to slice on a printer? Are you calling slic3r from a octopi, or some other web interface?

eyal0, 2017-01-12:
slic3r running from octopi, yes.  It's not as fast as doing it on a desktop but for simple STL files, it's a breeze to download from thingiverse and then upload it to octoprint.  Literally drag and drop to print a file.

CuraEngine is already built-in to octoprint but I wanted slic3r Prusa Edition (thanks for that!) because I read on the blog that it's the best for Prusa MK2.