prusa3d/PrusaSlicer #95: Compile problem - Win10 x64

playlet, 2017-01-13:
Hi!

Can someone tell me what is the working combination of perl (version of strawberry, x64/x86) and boost (version) needed to compile latest code under Windows 10 x64? I asked this twice now and can't seem to get an answer.

I use perl 5.24 and boost 1.59 for alexrj code and it works but I can't compile prusa3d version. Fails every time.

Please help.

Thanks, Ivan.

Sebastianv650, 2017-01-14:
I also have this problem since some weeks. The Only thing that we can do is to use a compiled version or (as i do) use a code base that's a little older that works. 
Some day before the next release it will be fixed i guess,be patient ;)

playlet, 2017-01-16:
Here's todays compile attempt:

<details>
  <summary>Perl response text</summary>
  <p>
Microsoft Windows [Version 10.0.14393]
(c) 2016 Microsoft Corporation. All rights reserved.

C:\WINDOWS\system32>cd c:\slicer\prusa\slic3r

c:\slicer\prusa\Slic3r>git pull
remote: Counting objects: 11, done.
remote: Compressing objects: 100% (11/11), done.
remote: Total 11 (delta 4), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (11/11), done.
From git://github.com/prusa3d/Slic3r
   a747a87e..e6b441ee  master     -> origin/master
Updating a747a87e..e6b441ee
Fast-forward
 xs/src/libslic3r/SupportMaterial.cpp | 38 ++++++++++++++++++++++--------------
 1 file changed, 23 insertions(+), 15 deletions(-)

c:\slicer\prusa\Slic3r>perl build.pl
App::cpanminus is up to date. (1.7042)
Class::XSAccessor is up to date. (1.19)
Devel::CheckLib is up to date. (1.07)
Encode is up to date. (2.88)
Encode::Locale is up to date. (1.05)
ExtUtils::MakeMaker is up to date. (7.24)
ExtUtils::ParseXS is up to date. (3.30)
File::Basename is up to date. (2.85)
File::Spec is up to date. (3.63)
Getopt::Long is up to date. (2.49)
IO::Scalar is up to date. (2.111)
Module::Build::WithXSpp is up to date. (0.14)
Moo is up to date. (2.003000)
POSIX is up to date. (1.65)
Scalar::Util is up to date. (1.47)
Test::Harness is up to date. (3.36)
Test::More is up to date. (1.302075)
Thread::Semaphore is up to date. (2.13)
Time::HiRes is up to date. (1.9741)
Unicode::Normalize is up to date. (1.25)
Could not find the MinGW directory at C:\dev\CitrusPerl\mingw32; skipping XML::SAX::ExpatXS (only needed for faster parsing of AMF files)
XML::SAX::ExpatXS is up to date. (1.33)
threads is up to date. (2.12)
Cleaning up build files
Cleaning up configuration files
cpanm (App::cpanminus) 1.7042 on perl 5.024000 built for MSWin32-x86-multi-thread-64int
Work directory is C:\Users\NI/.cpanm/work/1484582834.1728
You have make C:\Strawberry\c\bin\dmake.exe
You have LWP 6.15
Falling back to Archive::Tar 2.06
--> Working on ./xs
Entering C:/slicer/prusa/Slic3r/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Checking if you have Module::Build 0.38 ... Yes (0.4218)
Running Build.PL
Configuring C:/slicer/prusa/Slic3r/xs ... 'cc' is not recognized as an internal or external command,
operable program or batch file.

With @INC: "-Isrc/libslic3r", "-Isrc/glew/include", " -IC:\dev\boost_1_59_0"
With @LIBS: "-Lsrc/libslic3r", " -LC:\dev\boost_1_59_0\stage\lib", " -lboost_system-mgw49-mt-1_59", " -lboost_thread-mgw49-mt-1_59", " -lboost_log-mgw49-mt-1_59"
Created MYMETA.yml and MYMETA.json
Creating new 'Build' script for 'Slic3r-XS' version '0.01'
OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have Module::Build 0.3601 ... Yes (0.4218)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
[... 669 lines cut ...]
  Parse errors: No plan found in TAP output
t\18_motionplanner.t          (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t\19_model.t                  (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t\20_print.t                  (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t\21_gcode.t                  (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t\22_exception.t              (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
Files=21, Tests=0,  3 wallclock secs ( 0.09 usr +  0.28 sys =  0.37 CPU)
Result: FAIL
Failed 21/21 test programs. 0/0 subtests failed.
FAIL
! Installing ./xs failed. See C:\Users\NI\.cpanm\work\1484582834.1728\build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting

c:\slicer\prusa\Slic3r>
</p></details>

bubnikv, 2017-01-20:
Would it compile on Windows, if you set the environment variable
set SLIC3R_STATIC=1

Then Slic3r would be linked against a static Boost library. All our builds (my development build, the release Windows / Linux / OSX builds) are linked that way, so it may be, that the dynamic linking of Boost is broken.

playlet, 2017-01-20:
That worked, I can compile it now. Thank you very much sir! :)

playlet, 2017-03-08:
Hi!

There is some issue with compiling the latest code. What can I do to fix it?

Log:

c:\slicer\prusa\Slic3r>perl build.pl
App::cpanminus is up to date. (1.7042)
Class::XSAccessor is up to date. (1.19)
Devel::CheckLib is up to date. (1.07)
Encode is up to date. (2.88)
Encode::Locale is up to date. (1.05)
ExtUtils::MakeMaker is up to date. (7.24)
ExtUtils::ParseXS is up to date. (3.30)
File::Basename is up to date. (2.85)
File::Spec is up to date. (3.63)
Getopt::Long is up to date. (2.49)
IO::Scalar is up to date. (2.111)
Module::Build::WithXSpp is up to date. (0.14)
Moo is up to date. (2.003001)
POSIX is up to date. (1.65)
Scalar::Util is up to date. (1.47)
Test::Harness is up to date. (3.36)
Test::More is up to date. (1.302078)
Thread::Semaphore is up to date. (2.13)
Time::HiRes is up to date. (1.9741)
Unicode::Normalize is up to date. (1.25)
threads is up to date. (2.15)
cpanm (App::cpanminus) 1.7042 on perl 5.024000 built for MSWin32-x86-multi-thread-64int
Work directory is C:\Users\NI/.cpanm/work/1489010935.1300
You have make C:\Strawberry\c\bin\dmake.exe
You have LWP 6.22
Falling back to Archive::Tar 2.06
--> Working on ./xs
Entering C:/slicer/prusa/Slic3r/xs
Checking if you have Module::Build 0.38 ... Yes (0.4218)
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Running Build.PL
Configuring C:/slicer/prusa/Slic3r/xs ... Use of uninitialized value $_ in concatenation (.) or string at Build.PL line 117.
Slic3r requires the Boost libraries. Please make sure they are installed.

If they are installed, this script should be able to locate them in several
standard locations. If this is not the case, you might want to supply their
path through the BOOST_DIR environment variable:

    BOOST_DIR=/path/to/boost perl Build.PL

Or you may specify BOOST_INCLUDEPATH and BOOST_LIBRARYPATH separatly, which
is handy, if you have built Boost libraries with mutliple settings.

N/A
! Configure failed for ./xs. See C:\Users\NI\.cpanm\work\1489010935.1300\build.log for details.
The XS/C++ code failed to compile, aborting

playlet, 2017-03-15:
is anyone even reading this?

bubnikv, 2017-03-23:
There was a bug in the Build.PL script reporting wrong environment variables needed to make the static linking possible. The reporting has been fixed by https://github.com/prusa3d/Slic3r/commit/56f845f3c104ec51f6a5c6e5fb86d983ca9c9eb4

I have switched from Strawberry to own Perl distro compiled with Visual Studio 2013. The Strawberry should still compile, but there are some issues with the Strawberry 64bit (hangs on C++ exception followed by a croak), and our own binary distro of Perl seems to be working, in addition all the GUI components are pre-compiled and it runs 10% faster anyway.

Would you please try the VS2013 compilation? I have documented it here:
https://github.com/prusa3d/Slic3r/wiki/How-to-compile-Slic3r-Prusa-Edition-on-MS-Windows

Thanks.

bubnikv, 2017-03-23:
Closing. Please feel free to re-open this issue if you have any issues with the compilation on Windows. Thanks.

playlet, 2017-03-23:
I can't compile it using visual studio either. It keeps asking for boost libraries.

Can you help me debug the problem?

This is what I did:

I unpacked wperl64-5.24.0-2017-03-06.7z and wperl64d-5.24.0-2017-03-06.7z into c:\. Here I got c:\wperl64 and c:\wperl64d folders.
I installed Visual Studio Community 2013 into C:\Program Files (x86)\Microsoft Visual Studio 12.0.
I downloaded and extracted boost_1_63_0-msvc-12.0-64.exe into c:\local\boost_1_63_0
I downloaded and extracted tbb-2017_U5.7z into c:\local\tbb.

By runing C:\wperl64\cmdline.lnk, + perl build.pl I get this: output1.txt
(all paths are good in cmdline_init.bat)

By running C:\wperl64**d**\cmdline.lnk, + perl build.pl I get this: output2.txt

(I'm missing a path here to this thing: set WXDIR=D:\src-perl\wxWidgets-3.0.3-beta1 - where can I get these wxWidgets?)

I just can't get this to work. Are there any steps missing in the tutorial? It keeps asking for boost.

[output1.txt](https://github.com/prusa3d/Slic3r/files/865569/output1.txt)
[output2.txt](https://github.com/prusa3d/Slic3r/files/865570/output2.txt)

bubnikv, 2017-03-23:
You should be fine. I don't understand that.

If I were you, I would try to sprinkle prints into xs\Build.PL around the
lines 151 to 178 to understand what is going on. I would appreciate, if you
could do that.

playlet, 2017-03-23:
I'm willing to help you but please explain what "sprinkle prints" mean. :)
English is not my native language and I don't understand this.

playlet, 2017-03-23:
I've figured out what you meant by sprinkle prints. There is some problem with parsing $lib_ext.

This is what I get:

test_i9-3-C:\local\boost_1_63_0\lib64-msvc-12.0/boost_system*.aN/A
! Configure failed for ./xs. See C:\wperl64d\cpanm\work\1490298467.3928\build.log for details.
The XS/C++ code failed to compile, aborting

c:\local\slic3r>

If I put print here:

# Either static linking, or check_lib could not be used to find the boost libraries.
    **my $lib_prefix = 'boost_'; #was libboost_**
    PATH: foreach my $path (@boost_libs) {
        # Try to find the boost system library.
        my @files = glob "$path/${lib_prefix}system*$lib_ext";
		**print "test_i9-3-";
		print "$path/${lib_prefix}system*$lib_ext";**
        next if !@files;

bubnikv, 2017-03-24:
I think the script C:\wperl64xx\cmdline_init.bat did not invoke the visual studio path settings
If you print 
set
on the command line, you should get something like this:
Path=C:\Program Files (x86)\Microsoft Visual Studio 12.0\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files (x86)\MSBuild\12.0\bin\amd64;C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\BIN\amd64;C:\WINDOWS\Microsoft.NET\Framework64\v4.0.30319;C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\VCPackages;C:\Program Files (x86)\Microsoft Visual Studio 12.0\Common7\IDE;C:\Program Files (x86)\Microsoft Visual Studio 12.0\Common7\Tools;C:\Program Files (x86)\HTML Help Workshop;C:\Program Files (x86)\Microsoft Visual Studio 12.0\Team Tools\Performance Tools\x64;C:\Program Files (x86)\Microsoft Visual Studio 12.0\Team Tools\Performance Tools;C:\Program Files (x86)\Windows Kits\8.1\bin\x64;C:\Program Files (x86)\Windows Kits\8.1\bin\x86;C:\Program Files (x86)\Microsoft SDKs\Windows\v8.1A\bin\NETFX 4.5.1 Tools\x64\;
and so on.

It is the last line in the batch file
"C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\vcvarsall.bat" amd64

that seems to be failing I suppose.

bubnikv, 2017-03-24:
If you 

print "$lib_ext \n";

in the Build.PL, it should display

.lib

which is set at the line 15 if $cpp_guess->is_msvc is true, that means a visual studio compiler is detected.

I wonder whether you may have a MINGW compiler in your path as well, which is then being detected by the Perl.PL build system?

playlet, 2017-03-24:
if I run "C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\vcvarsall.bat" amd64 from cmd myself, I get nothing back in response. When I run smdline_init.bat I get this in the top of the window:
c:\local>set PATH=C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\AMD\ATI.ACE\Core-Static;C:\Program Files (x86)\ATI Technologies\ATI.ACE\Core-Static;C:\Program Files (x86)\QuickTime\QTSystem\;C:\adb;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Git\cmd;C:\Strawberry\c\bin;C:\Strawberry\perl\site\bin;C:\Strawberry\perl\bin;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SDKs\TypeScript\1.0\;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\AMD\ATI.ACE\Core-Static;C:\Program Files (x86)\ATI Technologies\ATI.ACE\Core-Static;C:\Program Files (x86)\QuickTime\QTSystem\;C:\adb;C:\Users\NI\AppData\Local\Microsoft\WindowsApps;c:\wperl64d\site\bin;c:\wperl64d\bin

answer to your next post:

print "$lib_ext \n"; - nope, I get ".a" - if I manually declare $lib_ext as .lib at line 15, then it continues and stops with the line asking for tbb path.

i have strawberry perl in c:\strawberry and boost_1_59_0 in c:\dev if that means anything here.

bubnikv, 2017-03-25:
I think you should get the strawberry Perl from your system path and make a
similar batch to open the command prompt with the strawberry path
preconfigured as I did for the visual studio Perl. Alternatively you may
reinstall strawberry, verify that it removes itself from the system path
and then install the zipped strawberry. That way you may install multiple
strawberries side by side.

playlet, 2017-03-28:
i managed to compile slic3r prusa after uninstalling strawberry perl from my windows pc.

i only needed to add sett tbb_... variables that are not included in the preconfigured perl zip file.

Thank you bubnikv! :)

closed 2017-03-28 by playlet

bubnikv, 2017-03-28:
Great! Thanks for testing.
I shall update the perl zips to include those paths. I hoped I already did.