prusa3d/PrusaSlicer #96: Adding option to switch infill and perimeters on top layer

Saij, 2017-01-14:
Hi,

would it be possible to implement an option to change the order of infill and perimeter only for the top most layers? It should result in better quality of flat surfaces.
At least that's what I encountered on my Anycubic i3. With infill before perimeters there weren't any bumps or sharp edges where the infill met the perimeter and so I'm currently printing with this setting.
But it is not so good for the first layer in terms of quality so I would appreciate such option to change it based on current layer?

Greetings
Saij

Sebastianv650, 2017-01-14:
If you print infill before the perimeter for top layers, will this not result in not 100% clean perimeters? The material that results in rough infill with the default print order is also there if you reverse it, so it should disort the perimeters somehow..

In any case, I see this only as a work around. With perfect esteps calibration, speed setting etc I'm also getting a perfect top solid layer without that.

JohnnyDeer, 2019-10-11:
Hi, this function is already implemented in few last versions of PrusaSlicer.

closed 2019-10-11 by JohnnyDeer