prusa3d/PrusaSlicer #97: Retina Support on OS X

kelchm, 2017-01-14:
### Version
1.31.6-prusa3d
### Operating system type + version
OS X 10.12

### Behavior
Slic3r does not render correctly on retina displays under OS X. In particular, all text is pixelated/blurry. 
[image]

Based on some preliminary research, wxwidgets should be able to support retina displays under OS X. Unfortunately simply editing the `info.plist` for the application did not solve the issue.

I plan to dig into this to see if I can get to the bottom of the issue and will send a pull request if I am successful. If anyone has any related information to share I would be grateful.

kelchm, 2017-01-14:
Refs alexrj/Slic3r#2888

It seems that some have had luck with 'Retinizer' for the text, though this has not worked for me so far.

kelchm, 2017-01-14:
This was simpler than expected.

The application `info.plist` did not have a valid `CFBundleIdentifier`.  Removing or setting the value for `CFBundleIdentifier` to any string and adding a `NSPrincipalClass` allows the text to be rendered in retina resolution.  The icons are still pixelated simply because the source images must be scaled. Unfortunately the Slik icons are not available in a higher resolution.

```.xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
  <key>CFBundleExecutable</key>
  <string>Slic3r</string>
  <key>CFBundleGetInfoString</key>
  <string>Slic3r Copyright (C) 2011-2016 Alessandro Ranellucci, (C) 2016 Vojtech Bubnik, Prusa Reseach</string>
  <key>CFBundleIconFile</key>
  <string>Slic3r.icns</string>
  <key>CFBundleName</key>
  <string>Slic3r</string>
  <key>CFBundleShortVersionString</key>
  <string>Slic3r 1.31.6-prusa3d</string>
  <key>CFBundleIdentifier</key>
  <string></string>
  <key>CFBundleInfoDictionaryVersion</key>
  <string>6.0</string>
  <key>CFBundlePackageType</key>
  <string>APPL</string>
  <key>CFBundleSignature</key>
  <string>????</string>
  <key>CFBundleVersion</key>
  <string>1.31.6-prusa3d</string>
  <key>CGDisableCoalescedUpdates</key>
  <false/>
  <key>NSPrincipalClass</key>
  <string>NSApplication</string>
</dict>
</plist>
```

bubnikv, 2017-01-27:
Thanks a lot! I will review it and integrate it into the following release.

litris, 2017-01-31:
It does not work for me.

I pasted the same text as you show and the text is pixelated in an imac 27 "5k late 2015.

`<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
  <key>CFBundleExecutable</key>
  <string>Slic3r</string>
  <key>CFBundleGetInfoString</key>
  <string>Slic3r Copyright (C) 2011-2016 Alessandro Ranellucci, (C) 2016 Vojtech Bubnik, Prusa Reseach</string>
  <key>CFBundleIconFile</key>
  <string>Slic3r.icns</string>
  <key>CFBundleName</key>
  <string>Slic3r</string>
  <key>CFBundleShortVersionString</key>
  <string>Slic3r 1.31.6-prusa3d</string>
  <key>CFBundleIdentifier</key>
  <string></string>
  <key>CFBundleInfoDictionaryVersion</key>
  <string>6.0</string>
  <key>CFBundlePackageType</key>
  <string>APPL</string>
  <key>CFBundleSignature</key>
  <string>????</string>
  <key>CFBundleVersion</key>
  <string>1.31.6-prusa3d</string>
  <key>CGDisableCoalescedUpdates</key>
  <false/>
  <key>NSPrincipalClass</key>
  <string>NSApplication</string>	
</dict>
</plist>

happydasch, 2017-02-01:
Fixed this issue by adding the following to info.plist:

```
<key>NSHighResolutionCapable</key>
<true/>
```

Complete content of info.plist:

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
  <key>CFBundleExecutable</key>
  <string>Slic3r</string>
  <key>CFBundleGetInfoString</key>
  <string>Slic3r Copyright (C) 2011-2016 Alessandro Ranellucci, (C) 2016 Vojtech Bubnik, Prusa Reseach</string>
  <key>CFBundleIconFile</key>
  <string>Slic3r.icns</string>
  <key>CFBundleName</key>
  <string>Slic3r</string>
  <key>CFBundleShortVersionString</key>
  <string>Slic3r 1.31.6-prusa3d</string>
  <key>CFBundleIdentifier</key>
  <string></string>
  <key>CFBundleInfoDictionaryVersion</key>
  <string>6.0</string>
  <key>CFBundlePackageType</key>
  <string>APPL</string>
  <key>CFBundleSignature</key>
  <string>????</string>
  <key>CFBundleVersion</key>
  <string>1.31.6-prusa3d</string>
  <key>CGDisableCoalescedUpdates</key>
  <false/>
  <key>NSPrincipalClass</key>
  <string>NSApplication</string>
  <key>NSHighResolutionCapable</key>
  <true/>
</dict>
</plist>
```

bubnikv, 2017-02-09:
The info.plist will be updated in the next official OSX build.

However, the anti-aliasing and Retina support for the 3D previews is more difficult. The anti-aliasing is a question of rebuilding the wxWidgets on the build server, but the Retina support is a nightmare, see the following links:

https://developer.apple.com/library/content/documentation/GraphicsAnimation/Conceptual/HighResolutionOSX/Introduction/Introduction.html
https://developer.apple.com/library/content/documentation/GraphicsAnimation/Conceptual/HighResolutionOSX/CapturingScreenContents/CapturingScreenContents.html
https://forums.wxwidgets.org/viewtopic.php?t=36684
http://stackoverflow.com/questions/12821144/how-to-draw-text-in-opengl-on-mac-os-with-retina-display

markaswift, 2017-03-09:
It would be awesome if this could be done, Slic3r is awesome, but it looks so 1990 on a retina screen :(

bubnikv, 2017-03-09:
That should be already in the 1.33.8 builds.

markaswift, 2017-03-09:
Hi @bubnikv, including icons?

bubnikv, 2017-03-09:
We don't have high resolution icons. We need a volunteer to paint them.

ArnieX, 2017-03-09:
Does it make sense before PrusaControl replaces slic3r? If so, how many icons are there? I might try to redraw them if it is not 100+ icons :D

bubnikv, 2017-03-09:
> Does it make sense before PrusaControl replaces slic3r? If so, how many
icons are there? I might try to redraw them if it is not 100+ icons :D

There is no plan to replace Slic3r with PrusaControl completely. Slic3r
will always offer much more controls for tweaking the slicing process while
PrusaControl is meant to be a bullet proof tool for beginners with little
knobs to play with.

ArnieX, 2017-03-09:
OK then I will have a look how many icons are there and will think about it.

bubnikv, 2017-03-09:
Great, thank you.

ArnieX, 2017-03-23:
Just a quick update. Author of the original icon set isn't offering hi-res version and is himself referring to Fogue Icon Set http://p.yusukekamiyamane.com which looks promising but I haven't had a chance to have a look yet.

markaswift, 2017-03-23:
They are truly shocking icons, 1980's? There are a ton of decent vector image sets out there, given how many icons are present in Slic3r, it really can't be a huge job to replace the assets, it would really give the UI a nice makeover too.

ArnieX, 2017-03-23:
Well I think Slic3r is over kill when it comes to icons. There are icons everywhere for almost any feature. There are many sets but many aren't free, and if they are free they always miss some important icons. It seem that it would be great to draw them from scratch just for Slic3r but that is something I can't do unfortunately.

markaswift, 2017-03-23:
Indeed, perhaps icon overdrive. That said, given the lack of time, a decent set of licensed icons could be had for next to nothing and would make the job far easier.

commit 4081d137 (2019-01-24): Mac: Implement Retina for the 3D scene, fix #97
commit d1c569dd (2019-01-24): Mac: Implement Retina for the 3D scene, fix #97
closed 2019-01-26 by bubnikv