prusa3d/PrusaSlicer #98: Unable to use preview beyond 25-30 rendered layers

VanessaE, 2017-01-16:
### Version

prusa3d fork, commit e6b441eea45029014ce928a7ffcccc78c906a63c

### Operating system type + version

Debian 8.6

### Behavior

I loaded the model below to check on an issue I was having with upstream, just to see if it's solved in this fork.  I changed a few settings to match what I last used to print the model, and sliced it.  When I click the "Preview" tab to check the sliced result, Slic3r UI hangs, and on the terminal, it throws hundreds of errors like this:

```
radeon: Failed to allocate virtual address for buffer:
radeon:    size      : 5869568 bytes
radeon:    alignment : 4096 bytes
radeon:    domains   : 2
radeon:    va        : 0x00000000fffd2000
```

...with the "size" field varying.  When I turn down the number of displayed layers (using only the slider on the right), this line repeats until I get to less than about 28 layers:

`radeon: The kernel rejected CS, see dmesg for more information.`

Looking at dmesg:

```
[  259.678404] radeon 0000:02:00.0: va above limit (0x0010056B > 0x00100000)
[  259.680024] radeon 0000:02:00.0: va above limit (0x0010056B > 0x00100000)
[  259.682500] [drm:radeon_cs_ioctl] *ERROR* Failed to parse relocation -12!
```

There are hundreds of the first two lines, with the first address varying (probably one such line for each of the errors Slic3r threw).  The last line is repeated each time I move the layers slider.

The video card is an R9 280X, using the open source Gallium driver as supplied by Debian.

Upstream works fine.

#### STL/Config (.ZIP) where problem occurs

[Flyer_1903_P_white_1.stl.zip](https://github.com/prusa3d/Slic3r/files/708125/Flyer_1903_P_white_1.stl.zip)

bubnikv, 2017-01-20:
This may be related to the changes I have done to the rendering code, namely I am sending the triangles and quads to the graphic card in small batches. 

You may try to effectively disable splitting the buffers to small pieces by editing 3DScene.pm, 
You will find following lines, influencing how big buffers are sent to the OpenGL driver.
                my $nvertices_this = ($nvertices > 4096) ? 4096 : $nvertices;
                my $nvertices_this = ($nvertices > 4095) ? 4095 : $nvertices;

Just replace them by 
                my $nvertices_this = $nvertices;
and the buffers will not be split.

The splitting above may be superfluous, but there is another place in the prusa3d fork, where the opengl buffers are split: In file 3DScene.pm, sub load_print_object_toolpaths, variable $alloc_size_max is currently set to 32MiB. Without this limit, Slic3r was not able to visualize large path previews and the OpenGL driver often gave up visualizing any OpenGL after that.

VanessaE, 2017-01-22:
I'm not too inclined to give that a try at this time.  I always seem to screw things up whenever I try a patch :smile:

bubnikv, 2017-02-17:
I did not receive any further response and I did not receive a similar issue from anybody else. Likely you have a buggy OpenGL driver. Please test with the official Linux AppImage. Please reopen this issue if you are still having this issue and you have more information for me to troubleshoot the issue.

closed 2017-02-17 by bubnikv