prusa3d/PrusaSlicer #106: M204 used instead of M201 when setting acceleration

nebbian, 2017-01-30:
### Version
1.31.6-prusa3d

### Operating system type + version
Mac OS X El Capitan 10.11.6

### Behavior
Setting the acceleration in the advanced tab sends M204 GCodes instead of M201.

To reproduce, set the acceleration in the advanced tab to 100 for the first layer, then 1000 for the default acceleration.  

[image]

The GCode produced uses M204 to set acceleration, instead of the more compatible M201.  Note, Repetier uses M204 to set PID values, not acceleration.

[image]

Here are some links describing the problem:
https://github.com/repetier/Repetier-Firmware/issues/290
http://reprap.org/wiki/G-code#M201:_Set_max_printing_acceleration

Is there a way to modify the printer settings so that it uses M201 instead of M204?

This isn't really a new request, more a bug fix.

Not that it really helps, but I've added the config file here:
[acceleration_config.ini.zip](https://github.com/prusa3d/Slic3r/files/738300/acceleration_config.ini.zip)

bubnikv, 2017-01-30:
Thanks, I know @lordofhyphens did some fixes for repetier on the main Slic3r. I will port them once I have the current code base released.

nebbian, 2017-01-30:
Thanks @bubnikv for the quick response 👍 

For anyone wanting a quick fix without needing another build of Slic3r, I've created this post processing script that seems to do the job:

`
#!/usr/bin/perl -i
#
# Post-processing script for changing default M204 acceleration control,
# to M201 which is compatible with Repetier firmware

use strict;
use warnings;

while (<>) {
    if (/^(M204\s+S(\d+))/) {
        printf "M201 X%d Y%d\n", $2, $2;
    } else {
        print;
    }
}
`

bubnikv, 2017-02-09:
To be merged:
https://github.com/lordofhyphens/Slic3r/commit/e0d8101627659ef2dcf1c49d5c9b512eb4e31a46
https://github.com/lordofhyphens/Slic3r/commit/885f27b8aea0df8be351825b9dd6061696f5edc9
https://github.com/lordofhyphens/Slic3r/commit/e56a29aaf3d9f67bfe9a0bae5c85356b3db4c40c

bubnikv, 2017-02-10:
Hopefully fixed by https://github.com/prusa3d/Slic3r/issues/106
Thanks to @lordofhyphens

closed 2017-02-10 by bubnikv