prusa3d/PrusaSlicer #107: Solid area split into unnecessary subsections

ceryen, 2017-01-31:
### Version
1.31.6-prusa3d-win64

### Operating system type + version
Windows 10

### Behavior
Use the default "0.20 mm NORMAL" profile, but disable "Ensure vertical shell thickness".  Slice the attached STL file and examine layers 3.2 and 3.4.  These two layers could be printed as a solid circle, but are instead split into multiple unnecessary subsections (10 extra trapezoids, 1 extra circle):

http://imgur.com/414g2pI.png

These unnecessary subsections cause the layers to be more likely to fail, particularly since the small trapezoids are printed over sparse infill.

If you enable "Ensure vertical shell thickness", the output changes slightly.  Layer 3.4 stays the same, but layer 3.2 actually becomes an almost solid circle with just one extra circle subsection.  It seems a little odd that layer 3.2 is almost solid, but the following layer is split into more subsections:

http://imgur.com/Cr5CTR9.png

The original Slic3r has similar issues.  Increasing "Solid infill threshold area" in the original Slic3r sometimes helps to combine the sections into one section, but that feature seems to be incredibly buggy and does not yield consistent results.  Changing "Solid infill threshold area" in Prusa Slic3r doesn't seem to have any impact.

#### STL/Config (.ZIP) where problem occurs
[Iron Man Coaster.zip](https://github.com/prusa3d/Slic3r/files/742673/Iron.Man.Coaster.zip)

ceryen, 2017-01-31:
I wonder if the subsections might be happening because of different extrusion widths for infill/solid infill/top solid infill.  If I set all the extrusion widths to be the same and leave "Ensure vertical shell thickness" off, layer 3.4 does become a single solid circle, but layer 3.2 still has the unnecessary subsections.

Overall, I'm not sure whether to characterize this as a bug, enhancement request, or working as intended.  Please feel free to do whatever you like with this issue =P

bubnikv, 2017-02-02:
> I wonder if the subsections might be happening because of different extrusion widths for infill/solid infill/top solid infill.

Exactly this is happening.

> layer 3.4 does become a single solid circle, but layer 3.2 still has the unnecessary subsections.

Slic3r recognizes a top solid infill, bottom solid infill, bottom bridging solid infill and normal solid infill. If these infills have the same properties, Slic3r merges them, which happens if you set the top infill extrusion width the same as the others. I believe the other case is caused by the bridging flow used over the sparse infill. This actually makes sense. It is very difficult to draw thin extrusions in the thin air over a sparse infill. Imagine 50 microns layer height.  The printer would not be able to close an infill with such extrusions, therefore a bridging flow is used.

I am marking this request as an enhancement, which I am already having on my quite a long TODO list. Ideally Slic3r would merge infill areas with somehow dissimilar properties, if one of them is small enough.

bubnikv, 2017-02-02:
Please let me know if you don't agree with the explanation given above and you think there is a real bug to be fixed ASAP.

ceryen, 2017-02-02:
The explanation above does make sense to me.  I agree with marking this as an enhancement and I don't believe that this needs to be fixed ASAP.

Actually, it seems like it might be difficult to decide whether two areas with dissimilar properties can be merged, so I wouldn't be too surprised if you decide this can't be reasonably addressed =P

ceryen, 2017-02-02:
Also, with a better understanding of what's happening, I see now that base Slic3r can generate similar output.  It might make sense for me to file a bug against base Slic3r instead.  Any thoughts/objections?

bubnikv, 2017-02-08:
I have no objections for you to enter a bug against the base Slic3r :-)

AFAIK there are already similar issues entered against the base Slic3r, if
you take your time, you will find them. It is more useful to add your
thoughts there.

closed 2026-09-29 by prusaslicer-bot[bot]