prusa3d/PrusaSlicer #108: Missing infill for portions of model

ghandye, 2017-02-01:
### Version
1.31.6-prusa3d-win64

### Operating system type + version
Windows 10.0.14393 x64

### Behavior
* _Describe the problem_
Slic3r is not generating infill for some portions of the model, leaving a single free-floating and unsupported top layer.

* _Steps needed to reproduce the problem_
Slice the attached STL file with the attached config. 
Switch to preview and examine layer Z=6.50, which is the last correctly formed layer.
Move preview layer up to Z = 25.90 and notice how the "back right" section of the model (next to one of the holes) is hollow with no infill.
Move preview layer up to Z = 26.10 and note how Slic3r attempts to lay a normal top-layer infill over open air in this corner.

* _Expected Results_
Infill should be generated in the portion of the model that is hollow.

* _Actual Results_
Infill is not generated in certain portions of the model (see blue circle)
[image]

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
[stl_and_config.zip](https://github.com/prusa3d/Slic3r/files/746164/stl_and_config.zip)

ghandye, 2017-02-01:
Did some more tinkering, and got the problem to go away by adjusting the "solid infill threshold area" to a lower value (i.e. decreased from default of 70 mm^2 to 5 mm^2).

Could this be a bug in the solid infill generation routine? Or is this expected behavior?

bubnikv, 2017-02-02:
Thanks for the report. I was not able to reproduce the bug with the current
source code. I believe it is a duplicate of the following:

https://github.com/prusa3d/Slic3r/issues/60
https://github.com/prusa3d/Slic3r/issues/79
https://github.com/prusa3d/Slic3r/issues/101

which has been fixed by

https://github.com/prusa3d/Slic3r/commit/17db5bff8d722369b648d41644782772f285ecd0

I hope to release a beta in the following days.

bubnikv, 2017-02-02:
Closing, not reproducible with the current source code. Feel free to reopen this issue, if you are able to reproduce this problem with the current source code (if you are able to compile Slic3r yourself from sources), or if you will be able to reproduce the issue once the next Slic3r is released.

closed 2017-02-02 by bubnikv