prusa3d/PrusaSlicer #109: Platter - update Goto undefined subroutine &AutoLoader::AUTOLOAD

alexxy, 2017-02-03:
### Version
1.31.6
### Operating system type + version
Linux

### Behavior

$ slic3r.pl 
Use of uninitialized value in subroutine entry at /usr/lib64/perl5/vendor_perl/5.24.1/Slic3r/GUI/Plater.pm line 159.
Platter - update
Goto undefined subroutine &AutoLoader::AUTOLOAD at /usr/lib64/perl5/vendor_perl/5.24.1/x86_64-linux-thread-multi/OpenGL.pm line 6178.

Version of dev-perl/OpenGL 0.700, However older one have similar behavior

bubnikv, 2017-02-11:
Did you compile it yourself? I guess you have the perl installation messed up.
Does our Linux binary work for you?

alexxy, 2017-02-12:
I compiled it localy from git master branch. Also i see same issue with original slic3r with its recent changes.

Your binary version works fine. Other perl packages also.

bubnikv, 2017-02-17:
I believe you have your Perl installation messed up. Closing this issue. Feel free to reopen it, if you don't agree with my conclusion.

closed 2017-02-17 by bubnikv