prusa3d/PrusaSlicer #110: Feature request: overhang compensation

Gobol, 2017-02-06:
Feature request: 

Printing overhanging wall typically comes out a little bit lower than designed (due to bridging or support clearance) it may be beneficial to raise it arbitrary no of layers (configurable) during slicing.

I have idea that faces with normals going toward bed [0,0,-1] could be raised by layer_height*x before mesh enters slicing engine. 

Any thoughts on this?

bubnikv, 2017-02-08:
Something like this certainly makes sense. I have it on my very long list already.

> Printing overhanging wall typically comes out a little bit lower than designed (due to bridging or support clearance)

It is actually not just the clearance which lets the threads to sag, but these threads are printed with a bridging flow in Slic3r, meaning these layers are a priory very thick compared to normal layers. For example, with the layer height of 0.1mm and a nozzle diameter of 0.4mm, the bridging layers are extruded with 0.4mm diameter threads, so the bottom layers over supports are 0.4mm high in Slic3r and sticking 0.3mm out of the bounds of the target geometry. We shall do something about it: 

1) Make the thickness of the bridging layers configurable.
2) Account for the high thickness of the bridging layers during the slicing process to meet the target geometry.
3) And account for the thread sag, which has probably the least effect of the three.

Gobol, 2017-02-08:
Thanks for your reply. Hope you'll find time to implement successful solution.

closed 2026-09-29 by prusaslicer-bot[bot]