prusa3d/PrusaSlicer #114: No support generated

Sebastianv650, 2017-02-09:
### Version
Creation date of this one

### Operating system type + version
Win10 64bit

### Behavior
No matter witch stl I choose and what options I set (overhang threshold, pattern, ..) in the support section, there is no support structure generated. I also checked the gcode output to see if it's only a visualisation thing - it isn't.

I hope it's not again a mistake intruduced by myself, but I don't think so because rest of slic3r is now compiling and running great.

bubnikv, 2017-02-09:
This is certainly weird. I am collecting a change log, I will release the beta binaries tomorrow.

Sebastianv650, 2017-02-10:
I repeated the support test with the new released beta, same situation. I'm using default settings for supports and tried multiple stl files - nothing. Realy strange..

bubnikv, 2017-02-10:
Would you please provide
1) the config
2) the stl?
Thanks

Sebastianv650, 2017-02-10:
Here they are:

[Example.zip](https://github.com/prusa3d/Slic3r/files/767459/Example.zip)

bubnikv, 2017-02-10:
Thanks, I can reproduce it now. It is something about your configuration,
that this slic3r does not like. I will look into it.

bubnikv, 2017-02-11:
The problem is in the setting of the first layer height (100% of the normal layer height). There is a bug in the new supports, which does not work with the percent value of the first layer height correctly. If you enter 0.2 into the first layer height field, the supports will work.

I will fix it soon. Thanks for the bug report.

bubnikv, 2017-02-11:
Fixed by https://github.com/prusa3d/Slic3r/commit/8d6acd2aec6f2ac9129d9ba55bf0f51fb6ffd3b7

closed 2017-02-11 by bubnikv