prusa3d/PrusaSlicer #115: 1.33.3 release fails to start on Windows 10

ceryen, 2017-02-10:
### Version
1.33.3

### Operating system type + version
Windows 10

### Behavior
Attempting to start the 1.33.3 beta release fails with:
The application has failed to start because its side-by-side configuration is incorrect. Please see the application event log or use the command-line sxstrace.exe tool for more detail.

The application event log contains the following information:
Activation context generation failed for "D:\3D Printer\Slic3r-1.33.3-prusa3d-win32-full-201702101112\slic3r.exe".Error in manifest or policy file "D:\3D Printer\Slic3r-1.33.3-prusa3d-win32-full-201702101112\slic3r.exe" on line 3. The value "1.33.3" of attribute "version" in element "assemblyIdentity" is invalid.

#### STL/Config (.ZIP) where problem occurs
N/A

ceryen, 2017-02-10:
I tried both win32 and win64 versions in case it made a difference, both gave the same errors

3d-gussner, 2017-02-10:
Same here: Eventlog ID 63
slic3r-console.exe" on line 3. The value "1.33.3" of attribute "version" in element "assemblyIdentity" is invalid.

ceryen, 2017-02-10:
The problem seems to be the new slic3r.exe binary.  Using the old slic3r_gui.bat allows slic3r to start up fine.  Here are the contents of the old slic3r_gui.bat in case someone needs it:

@start "Slic3r" "%~dp0\wperl.exe" "%~dp0\slic3r.pl" %*

bubnikv, 2017-02-10:
Yeah, I will replace the binary package right away. The problem is in the manifest, which is produced by the build script. Windows does not swallow a version 1.33.3, it has to be 1.33.3.0

bubnikv, 2017-02-10:
Please get the updated version from the release page. The problem concerned the win32 / win64 binary builds only.

closed 2017-02-10 by bubnikv