prusa3d/PrusaSlicer #116: Crashes when trying to disable variable layer height feature

CReimer, 2017-02-10:
### Version
1.33.3

### Operating system type + version
Arch Linux

I tried to disable the variable layer height feature, as it has no real use to me without an automatic mode.

When I try to disable the variable layer height feature Slic3r crashes.
When I disable it in the config Slic3r refuses to load the config file with the following error: 

"I was unable to load the selected config file: Can't call method "Disable" on an undefined value at /usr/share/perl5/vendor_perl/Slic3r/GUI/Plater.pm line 1639."

bubnikv, 2017-02-10:
Thanks for the report. I did not test on Linux. It seems to be some kind of race condition specific to Linux.

By the way, why is the manual layer height of no use to you? Did you find any of the automatic layer height generators useful? On which models?

CReimer, 2017-02-10:
Actually I never tried these generators. I just assumed they do everything on their own and I don't have to worry about it. This sounds a bit like they don't do that?

bubnikv, 2017-02-11:
> I just assumed they do everything on their own and I don't have to worry about it. This sounds a bit like they don't do that?

They do that, but for all but the simplest geometries they produce something you don't want. Namely, often the profiles are jagged and often they find a triangle in each layer, which dictates the thinnest layer allowed. I have addressed that in my release notes.

It is much more complicated to do the automatic variable layer height right. For example, you don't want to have too thin layers on overhangs, you don't want the thin layers to be triggered by features which are not visible, you want the transitions to be smooth etc. My implementation of the preview makes it easy to achieve nice surfaces of prominent features of complex models, for example of the Benchy's roof.

bubnikv, 2017-02-11:
I can reproduce the crash with our Linux binaries and the ArchLinux in the VirtualBox.

bubnikv, 2017-02-11:
A different toolbar type is used on WIndows than on OSX & Linux. Some of the code for OSX & Linux is missing. I will fix that. Thanks for a bug report.

bubnikv, 2017-02-12:
I just verified that it crashes the same way on OSX.

Wurstnase, 2017-02-13:
I just compiled the latest verions but I get:
```
Use of uninitialized value in subroutine entry at /home/nico/Tools/Slic3r_Prusa/lib/Slic3r/GUI/Plater.pm line 167.
Can't locate object method "variable_layer_height" via package "Slic3r::Config" at /home/nico/Tools/Slic3r_Prusa/lib/Slic3r/GUI/Plater.pm line 1765.
```
when trying to execute it.

bubnikv, 2017-02-13:
> I just compiled the latest verions but I get:

Most likely the C++ compilation did not go through and you are running the new Perl code against the old C++ code.

Wurstnase, 2017-02-13:
Is there something like `make clean`?

Wurstnase, 2017-02-13:
Ok. Got it to run. Only the first part is shown in the terminal. 
```
Use of uninitialized value in subroutine entry at /home/nico/Tools/Slic3r_Prusa/lib/Slic3r/GUI/Plater.pm line 167.
```
But it starts.

I had the old starting script from alexrj `eval $(perl -Mlocal::lib)`. Without everything is ok again.

bubnikv, 2017-02-15:
Fixed by https://github.com/prusa3d/Slic3r/commit/fd54956d708436bd4a9b15c951d4daa42498abb4

closed 2017-02-15 by bubnikv