prusa3d/PrusaSlicer #118: Unstable first object layer with raft

freiser77, 2017-02-10:
### Version
1.33.3-prusa3d

### Operating system type + version
macOS Sierra

### Behavior
* Slice an 40x40x40 cube (any model will do), default options working for your printer and set "raft layers" to 10
* The first layer of the cube sitting atop the raft will have a smaller extrusion width (not only in preview, but also verified in print), leading to a stringy bottom side.
* Screenshot from preview is attached.

#### STL/Config (.ZIP) where problem occurs
[Archiv.zip](https://github.com/prusa3d/Slic3r/files/767453/Archiv.zip)

bubnikv, 2017-02-13:
The current (and the old) raft respects the "Contact Z distance" at the support page. If you set the distance to 0.001, then it will print the object directly above the raft.

Also please note that the object's 1st layer is printed with a bridging flow and you should enable the "detect bridging perimeters" feature to have the perimeters be printed with a bridging flow as well. Whether that is a good approach is questionable, but it does not differ between the previous and the current Slic3r.

[image]

bubnikv, 2017-02-13:
Does this solve your issue?

freiser77, 2017-02-13:
@bubnikv: It is true, that it is not different from the previous version. The problem is not fixed with "Contact Z distance" either. I want it to separate from the raft easily. Therefore I need the 0.2 mm "Contact Z distance". The problem I meantioned is, that the extrusion is so little, that the bottom surface of the object does not stick together, but is an array of single filament strings. Exactly as displayed in the preview tab.

The "bridging flow" parameter is still 1 as bridging is working fine with that.
Do you suggest me to raise the bridging flow to account for that special case of rafts?

bubnikv, 2017-02-13:
> I want it to separate from the raft easily. Therefore I need the 0.2 mm "Contact Z distance". The problem I meantioned is, that the extrusion is so little, that the bottom surface of the object does [...]

I don't understand. The preview tab, at least for me and for the z gap 0.001 displays the 1st object layer touching the raft.

I believe one shall separate the raft settings from the support settings as the raft needs different separation tactics from the normal supports, but this will not be done in this release as it requires a lot of evaluation from our side.

freiser77, 2017-02-15:
@bubnikv: Up until now, I understood the z gap as necessary to separate the support/raft from the part. A too small value causes the part and the raft to fuse together. Am I correct?

bubnikv, 2017-02-15:
> Up until now, I understood the z gap as necessary to separate the support/raft from the part. A too small value causes the part and the raft to fuse together. Am I correct?

That's correct. Without the gap, the object would fuse to the raft. One probably needs a separate settings for the raft parameters as there are different requirements to it than for the normal support gap. But that will not be done in this release.

bubnikv, 2017-02-17:
I was testing the raft yesterday and there is something to be desired.

The current "bridging over support" extrusion placement works to some extent for overhangs, but it does not work well for long horizontal features, like a raft or models displayed in here:
https://github.com/prusa3d/Slic3r/issues/102

First, as discussed here https://github.com/prusa3d/Slic3r/issues/102, for the horizontal features to be printed nicely, one has to squash the horizontal overhang against the support top surface. This is particularly needed for ABS prints which curl up, and for object over raft.

Also there is an issue with the current bridging algorithm, which does not correctly fuse the bridging infill with the bridging perimeters:
https://github.com/prusa3d/Slic3r/issues/134

FireflyNJ, 2017-09-19:
I am having the same problem, the spacing between the raft and the project is fine, but for some reason the first layer density changes when the raft is turned on.

first layer of project, this looks nomal
[image]

First layer of project with raft on, notice the increase of space that you can see the raft under it.
[image]

I can not find any settings that changes this.

bubnikv, 2021-02-24:
Implemented with #6003 and aee136c

Object over raft is no more printed with bridging flow, but with a normal flow.
Gap between the raft and 1st object layer is configurable independently from the support gap.

closed 2021-02-24 by bubnikv