prusa3d/PrusaSlicer #119: Layer Editing: Suggestions for better usability

MartinMajewski, 2017-02-10:
### Version
1.33.3

I like the direction the development goes with the layer height editing. However, there are several suggestions I want to make for further consideration.

Sharp steps in layer height changes would be useful. As seen in the screenshot provided the top end of my adjustment is always smoothed out. The last couple of layers gets higher again, which is not what I want. In fact, the last layer (the topmost it is) is 0.12 mm, the second last is 0.09 mm, and from then all other layers are at the minimum of 0.07 mm until the bumpy gradient starts. That is the best I could get with clicking, shift+clicking, clicking and so forth.

With the current control scheme, it is not easily possible to adjust it correctly. On the one hand, there is some smoothening/interpolation going on internally. On the contrary, moving the curvature around by digitally clicking and scrolling will always be guesswork depending on luck and persistence.
So besides the request for removing the interpolation for the first and last layer, I would also suggest the following:

Revamp the control scheme in a way that provides something like splines with control points and tangent lines. Look at vector drawing tools like Illustrator or Affinity Designer, or any other tool that uses splines to interpolate a function. The control points can have characteristics for interpolation like a discrete or continuous interpolation. Setting points onto a canvas is a far easier procedure than doing analog sculpting with a PC-mouse. With the current Layer Editor, I feel a little bit like I would use Mudbox inside slic3r. :-D

Best wishes,
Martin
[image]

MartinMajewski, 2017-02-10:
Nevertheless, the current control scheme fits perfectly with the overall slic3r UI madness, where nothing is controlled the same way nor controlled within a constant level of detail, and everything is scattered around, hidden, behind a context menu, or showed as a popup menu. :-P

landodragon141, 2017-02-10:
👍 :-)

bubnikv, 2017-02-11:
Martin,

would you please provide a config.ini and export an AMF file with the layer edited the way it is on the screenshot? The top most layer should certainly be editable. If it is not, then it is a bug.

> Revamp the control scheme in a way that provides something like splines with control points and tangent lines.

I was surely thinking about it. It really depends on the use case of the tool. If you have a complex decorative object, the current UI is IMHO very good in allowing you to have a quick control over print detail and smoothness of the transitions. At least our prints we have done internally supports this.

> Look at vector drawing tools like Illustrator or Affinity Designer, or any other tool that uses splines to interpolate a function. The control points can have characteristics for interpolation like [...]

In my experience working with splines is cumbersome. It may be good for technical work where you want to achieve a prescribed profile. But most of the time the variable layer height will be used to make the top surfaces of 3D benchies and tree frogs nicer, and for this purpose the current UI is adequate. 

I am open for discussion. Let's wait for more feedback.

> Nevertheless, the current control scheme fits perfectly with the overall slic3r UI madness, where nothing is controlled the same way nor controlled within a constant level of detail, and everything is scattered around, hidden, behind a context menu, or showed as a popup menu. :-P

This is an open source project. You are most welcome to join the development and polish the UI, if you have the time and abilities.

bubnikv, 2017-02-15:
> The last couple of layers gets higher again, which is not what I want. In fact, the last layer (the topmost it is) is 0.12 mm, the second last is 0.09 mm, and from then all other layers are at the [...]

This has been addressed by the following change:
https://github.com/prusa3d/Slic3r/commit/4d739c41a00250508b9c31c8e66b8f59e505ce2e

bubnikv, 2020-02-05:
We will not implement spline controls. A bug mentioned was fixed.
The sharp layer transitions may newly be edited with the layer height table at the object list.
Closing.

closed 2020-02-05 by bubnikv