prusa3d/PrusaSlicer #120: Raft under object is missing

freiser77, 2017-02-10:
### Version
1.33.3-prusa3d

### Operating system type + version
macOS Sierra

### Behavior
* Slice any model with 10 raft layers and "contact z-distance" set to zero.
* When looking at the preview there is no raft under the object, leaving it float mid-air.
* Screenshot is attached

#### STL/Config (.ZIP) where problem occurs
[Archiv.zip](https://github.com/prusa3d/Slic3r/files/767462/Archiv.zip)

bubnikv, 2017-02-15:
This is similar to https://github.com/prusa3d/Slic3r/issues/122 IMHO.

freiser77, 2017-02-15:
This may be, yes.

bubnikv, 2017-02-15:
This bug was specific to soluble interfaces or very thin suport interface gaps.

Fixed trimming of support volumes by objects for the soluble supports and for the raft contact layer over soluble supports.

https://github.com/prusa3d/Slic3r/commit/d2d7c5beade5566111beca79bb3b072f67f6e8c6

closed 2017-02-15 by bubnikv