prusa3d/PrusaSlicer #121: Crashing when entering "layer editing" mode in prusa 1.33.3 build

SurfingSpider, 2017-02-10:
### Version Prusa 
1.33.3

### Operating system type + version
win64

Intel hd graphics 4600 (onboard GFX card)
20gb ram
intel i5

### Behavior

When I try to use the new layer height editing function (only when in the 3d tab) it crashes straight away. If I go through the console it gives me the following error.

Died at //d10-file-01/personal$/USERNAME/Desktop/Slic3r-1.33.3-prusa3d-win64
-full-201702101559/Slic3r-1.33.3-prusa3d-win64-full-201702101559/lib/Slic3r/GUI/
3DScene.pm line 1184.

#### STL/Config (.ZIP) where problem occurs

I tried several STL's simple and complicated.

bubnikv, 2017-02-11:
The OpenGL drivers seem to have a problem with the shaders. I am using a very old standard, which the very most of old and new computers should implement.

Would you please try following:

1) Edit the file Slic3r-1.33.3-prusa3d-win64-full-201702101559/lib/Slic3r/GUI/
3DScene.pm and remove a comment mark (#) at the start of the line 898
This is the line: 
#    print $info if $info;

2) Run slic3r using the command slic3r-console.exe --gui

This will show the error report given by the graphics card driver about what it thinks about the OpenGL shaders it likely was not able to compile.

Thanks

SurfingSpider, 2017-02-11:
I had run a grf driver update and fixed the problem just before you answered. It is working now. Thank you for your help

closed 2017-02-11 by SurfingSpider