prusa3d/PrusaSlicer #122: 1.33.3: skirt can become a few kilometers large, and is not always printed first

DrLex0, 2017-02-11:
### Version
1.33.3

### Operating system type + version
Mac OS X 10.11.6

### Behavior
* When using more than one raft layer, the skirt extends a few kilometers outside the print bed. When using support material without a raft, the skirt is not printed as the very first thing.
* Example: use attached STL file with attached config.
* Expected results: the skirt is printed around the object and stays on the print bed. The skirt is always printed first.
* Actual Results: the skirt takes various unexpected shapes and runs off the print bed towards the other side of town. When disabling raft layers, the support material is printed before the skirt.

No raft:
[image]
One raft layer:
[image]
Two raft layers:
[image]

#### STL/Config (.ZIP) where problem occurs

[ConfigAndSTL.zip](https://github.com/prusa3d/Slic3r/files/768614/ConfigAndSTL.zip)

bubnikv, 2017-02-13:
> When using support material without a raft, the skirt is not printed as the very first thing.

Are you sure about that? It seems to me that the skirt is printed as the first thing, if the raft is disabled.

DrLex0, 2017-02-13:
> Are you sure about that? It seems to me that the skirt is printed as the first thing, if the raft is disabled.

Yes, attached is a GCode file with the .ini that generated it. The support material is printed before the skirt, unlike in previous versions.

[RaftVsSkirt.zip](https://github.com/prusa3d/Slic3r/files/770567/RaftVsSkirt.zip)

Saij, 2017-02-13:
I can also say that the support is printed before the skirt and the model.
For me it seems that the support is always a layer below the rest of the model. GCode.ws and Pronterface show the support on a seperate layer than the rest of model (so it is always support, then model, then support, then model and so on).

Tested yesterday with model http://www.thingiverse.com/thing:986652
Standard Prusa settings, 100% infill, support enabled and no raft.

nebbian, 2017-02-15:
I noticed the same thing this morning regarding the skirt being printed after the support (and possibly brim).  This is different to all previous versions of Slic3r that I've used.  
Perhaps the first layer height isn't being applied to the support and brim?

bubnikv, 2017-02-15:
I have fixed some of the raft issues (missing first layer raft), but the
large raft for the particular stl and ini is still there. I suspect the
data gets mangled when transitioning the C/Perl boundary, which is really
difficult to debug. I will likely rather rewrite this piece of code to C++.

bubnikv, 2017-02-15:
Fixed the order of a skirt, raft, support & object on the 1st layer. This is a new issue after I switched the order of support and object extrusions being printed at the same Z height. By the way, are there pros / cons to printing the support before / after an object? My gut feeling is, it is better to print the support first due to the overhangs. https://github.com/prusa3d/Slic3r/commit/35490216a773a39b293db5c9e7d4cf9e2d1645a3

Fixed (hopefully) the issue with the skirt sticking out of the print bed by re-writing the suspicious part from Perl to C++.
https://github.com/prusa3d/Slic3r/commit/3d11d1aebfee4fc1a592832b60ed142874d1ada7

And finally, I have fixed a handful of bugs in the rafts, which were again introduced with the new C++ supports.
https://github.com/prusa3d/Slic3r/commit/f5e4026aeee5ae80d2084dd67801eb6956a2f350

closed 2017-02-15 by bubnikv

DrLex0, 2017-02-15:
> By the way, are there pros / cons to printing the support before / after an object? My gut feeling is, it is better to print the support first due to the overhangs.

It doesn't really matter I think, as long as the support is there when the next layer is printed on top of it… But intuitively I also prefer support first.

bubnikv, 2017-02-15:
> But intuitively I also prefer support first.

Good, because the old slic3r did it vice versa, the object first.