prusa3d/PrusaSlicer #123: Volumetric slope limits cause perimeters to be messed up

DrLex0, 2017-02-11:
### Version
1.31.6 and 1.33.3

### Operating system type + version
Mac OS X 10.11.6

### Behavior
I'm really interested in the autospeed feature because the largest loss in print quality I'm currently having, is due to sudden changes in extruder pressure that cause either ugly blobs on outer perimeters, or under-extrusion. Some of my filaments are very slow to react to pressure changes. I enforced speeds and extrusion widths for inner and outer perimeters to similar values to combat this, but this still fails when a perimeter happens to be around the threshold for switching between small and regular perimeter speed: the outer perimeter suddenly speeds up and under-extrudes. (I believe this is an issue in itself: all perimeters of a given contour should always be printed at the same speed, but that's a different story.)
The whole point of autospeed is to make these kind of tweaks unnecessary. However:

* When printing certain relatively simple models and imposing volumetric slope limits, the perimeters of the model are printed as a jumbled mess.
* To reproduce: slice the attached STL file with the attached settings. Actually, any settings will do, if you configure non-zero values for the volumetric slope limits.
* _Expected results_: it produces a correct GCode file.
* _Actual Results_: it produces a mess.

Without volumetric limits:
[image]

With volumetric limits:
[image]

#### STL/Config (.ZIP) where problem occurs
[ConfigAndSTL2.zip](https://github.com/prusa3d/Slic3r/files/768993/ConfigAndSTL2.zip)

bubnikv, 2017-02-15:
The volumetric slope is an experimental feature, which is known to break on tiny moves, for example on the gap fill. Please don't use it for now.

DrLex0, 2017-07-05:
I blatantly ignored your advice and used a positive slope limit on a print that suffered from under-extrusion due to a sudden transition from tiny gap fill to perimeters, and it worked pretty well. This makes me wish this feature would work more reliably.
I did see some strange things in that print, like the speed dropping to near zero sometimes and then slowly crawling up again, but this had no consequences aside from slowing down everything.

Also to make this truly useful, the limits (or any autospeed whatsoever) must not be applied to bridges. Bridges usually only work well at a specific speed, and it is better to have a bit of under-extrusion on a bridge than deviating from that optimal speed.

Sebastianv650, 2017-07-05:
Two notes:
- What you do with this feature is just limiting the acceleration. You get the same when you lower the acceleration in the FW. Why do it the complicated way in Slic3r, especialy when it's not stable?
- Have you tried LIN_ADVANCE? It should eleminate the need of this feature completely (except Ninjaflex and such I guess - not tested) Link for informations: http://marlinfw.org/docs/features/lin_advance.html

bubnikv, 2017-07-05:
Due to the constant acceleration trapezoidal profiles that the Marlin
firmware (and any grbl derived relative) implements, there is inherent
pumping of the extruder pressure at the transitions between the linear
segments. The linear advance improves on it, but it cannot completely
eliminate it. The idea of the volumetric slope limits is to decelerate soon
enough when printing visible paths, so that the pressure in the nozzle is
already stable.

We are having some stringing issues with our bowden multi-material setup.
It is clearly seen, that the strings get drawn during a travel succeeding
an infill. This is due to the high pressure in the bowden created during
the infill, which gets released during the travel. The idea of the pressure
control in the slicer is to control these effects.

DrLex0, 2017-07-05:
I'm using a different firmware (Sailfish). It does have both acceleration and JKN advance. However, as far as I can determine and as @bubnikv states, JKN is only really effective in compensating the effects of accelerated print moves. Quote from Sailfish documentation: _“In truth, some of these issues are best addressed by using an acceleration-aware slicer or gcode post-processor.”_

The advance algorithm fails badly in compensating for extreme changes in extruder pressure like going from gap fill to perimeter in a few milliseconds. It also fails to compensate for resuming after a long retracted move.
I have tried tweaking acceleration and JKN parameters, but this always results in a mess if I deviate too much from the typical values which seem to be a compromise. I may be able to improve upon one problem, but it worsens other problems. It just cannot handle both step-like responses and more subtle changes at the same time.

Moreover, imposing the same low acceleration to the extruder indiscriminately, will mess up bridges as I noted before. Some moves simply need to be abrupt. Plus, I only really need this kind of tweak for specific prints with specific filaments. I don't want to slow down all my prints or change firmware parameters all the time.

gnydick, 2019-01-14:
instead of waiting for people to read these comments from a year and a half ago, how about just disabling it, not even marking it as experimental, given, 99.999% of prints will have small moves? Seems pretty silly to even allow it to be used.

bubnikv, 2019-01-15:
> Instead of waiting for people to read these comments from a year and a
half ago, how about just disabling it, not even marking it as experimental,
given, 99.999% of prints will have small moves? Seems pretty silly to even
allow it to be used.

That is a great idea.

bubnikv, 2019-01-29:
Disabled the experimental pressure equalizer with 57bd0889c384b4a2420adbfa306de8c8d191b9b4.
Closing.

closed 2019-01-29 by bubnikv