prusa3d/PrusaSlicer #124: Add more info about Perl and Wx version

Eriobis, 2017-02-13:
### Version
version_1.33.3-61f9414b0963ff5c6f0087df10434beca28ced9d

### Operating system type + version
Ubuntu 16.04
Windows 10

### Behavior
I installed the latest Perl Strawberry
I got to install the boost lib
It keeps failing at building Xs,
I tried to install CPAN -i Wx and the wx module keeps failing, I beleive it's the source of the problem.

[image]

Can you provide info on what tools version you guys are using ??? I'm having the same problem on my linux box, this is a little bit frustrating. I know it may be a perl problem. I would like to know how you guys do it :) ?

Thanks a lot

bubnikv, 2017-02-13:
Perl is a mess. There are different issues with Windows / Linux / OSX. I have a build system set up on all three platforms, where some Perl packages are installed manually. After that I am forcing myself to forget the pain.

Strawberry Perl 64bit has an issue with exception reporting from a C++ code. That is the 22_exception.t, which hangs on a buggy Strawberry Perl 64bit. As I have applied a lot of patches to Slic3r to avoid using the C++ exceptions together with Perl Croak, You may compile Slic3r on Strawberry Windows 64bit with a SLIC3R_HAS_BROKEN_CROAK=1 environment set, then the 22_exception.t will be ignored and a check will be compiled into Perl to die hard on C++ & Perl croak at least to detect such combination. Our Slic3r 64bit is built with such settings.

bubnikv, 2017-02-13:
I agree I shall document these caveats and tricks.

Saij, 2017-02-13:
I feel with you ^^ Have also the Perl trouble ^^

landodragon141, 2017-02-14:
I salute you and your efforts to rewrite in C++. Perl is a huge pain for multi platform.

bubnikv, 2017-03-23:
Those issues should be gone with the pre-built perl and libraries:
https://github.com/prusa3d/Slic3r/wiki/How-to-compile-Slic3r-Prusa-Edition-on-MS-Windows