prusa3d/PrusaSlicer #125: Wrong version when build from source

murdock62, 2017-02-13:
### Version
1.33.3

Just one little newbie question.

Why when i build from source the version in the GUI is 1.31.6 ?

Many thanks for your help !

bubnikv, 2017-02-13:
We have an internal build setup, which gets the version from a git label and a git checkin number and then this number is integrated wherever needed. This way I don't have to put the version number manually.

murdock62, 2017-02-13:
Ok so after build my GUI display a wrong version number but the Build is the "true 1.33.3" ?

Many thanks again and sorry for my bad english !

bubnikv, 2017-02-13:
Your English is just fine. 

The build script uses the results of 
git describe

which for the current master returns
version_1.33.3-4-gce8973b

Then the build script just trims the version_ prefix and you have the version number.

Closing this issue. Enjoy.

closed 2017-02-13 by bubnikv

murdock62, 2017-02-13:
Many thanks !