prusa3d/PrusaSlicer #126: Weird/Slow movements after finishing skirt

Itox001, 2017-02-13:
### Version
Slic3r Prusa Edition 1.33.3

### Operating system type + version
Windows 7 64 bit

### Behavior
When slicing any object with skirt, after the skirt is done the printer head moves very slowly and with weird noises. This does not happen in 1.31.6. I checked out the gcode and in fact, there's 1000 lines of code of very very small movements, with a feedrate set in between each one. It looks like this: 

...G1 X96.536 Y89.827 E19.96997
G1 X96.791 Y89.319 E20.00730
G1 X97.085 Y88.840 E20.04422
G1 X97.423 Y88.384 E20.08155
G1 X97.798 Y87.961 E20.11868   //End of skirt
G1 F7200                                     //Start of weird travel moves
G1 X97.844 Y87.913 E20.08718
G1 F7200
G1 X97.849 Y87.909 E20.08418
G1 F7200
G1 X97.853 Y87.904 E20.08118
G1 F7200
G1 X97.858 Y87.900 E20.07818
G1 F7200
G1 X97.862 Y87.895 E20.07518
G1 F7200
G1 X97.866 Y87.891 E20.07218
G1 F7200
G1 X97.871 Y87.886 E20.06918... //1000 lines more of this

[image]

The weird movements correspond to the travel from the last point of the skirt (highlighted in yellow) to the point where it travels from the skirt to the object. I think I've seen it happen in the middle of printing an object too, when following the weird curved travel moves generated by Avoid Crossing Perimeters (I think). If it's needed, I can probably find the gcode where this happens in another object. 

Attached is the config, stl of the box in screenshot (hapens in every part, though), and gcode.
[SkirtIssue.zip](https://github.com/prusa3d/Slic3r/files/772381/SkirtIssue.zip)

nebbian, 2017-02-15:
I noticed the same thing this morning.  I could tell that it was ripping through the GCode due to the percentage complete value on the LCD increasing quickly while the head was slowly stuttering.
Version 1.33.3

bubnikv, 2017-02-15:
I see it, it is the wipe move.

@Itox001 @nebbian by the way, which printer & firmware?

bubnikv, 2017-02-15:
@lordofhyphens It is a general bug on extrusion of closed loops. The closed loop gets decimated to a reasonable resolution, but not the wipe move.

closed 2017-02-15 by bubnikv