prusa3d/PrusaSlicer #127: Partial bridging for layers with holes

Ghrem, 2017-02-13:
### Version
1.33.3-prusa3d-win64

### Operating system type + version
Windows 10

### Behavior
No bridging is done for layers printed over deadspace that have a hole in them.
Print a 20mm nut with a 4mm hole over the top, the 4mm hole prints into dead space with no attachments. http://www.thingiverse.com/thing:2106054 for example.

Expected partial bridging from outside edges to edge of interior hole at 0º, at the minimum, optionally a 2nd bridging at 180º for four corner support of interior hole on next layer.
http://forums.reprap.org/read.php?263,747848

http://forums.reprap.org/file.php?263,file=90485
http://forums.reprap.org/file.php?263,file=90486

Iam3delight, 2017-02-13:
You can't bridge to a non existing 'hole' in mid air.... Just turn supports on.

If you try to bridge:
Hole first - As the holes internal perimeters have to be extruded in mid air it will go wrong.
Bridges first - As the holes internal perimeters will not have been created yet it will fail.

Simple laws of physics!  The only option is to turn on supports...

Ghrem, 2017-02-14:
Really? You can't bridge on Layer 20 (file=90485) to the left/right (or whatever orientation) of the hole on Layer 21 and just not place bridging in the center where the hole is on 21?... Your "laws of physics" statement is really stupid, btw.  It's just a damn bridge without the center printed.

Ghrem, 2017-02-14:
[image]

Like this.... physics be damned. lmfao

Ghrem, 2017-02-14:
Simply asking for this to be "automatic" as it can be achieved by manually adding a .2mm surface to the left/right of the hole in the nut space (on this example object) as seen here:
[image]

I'd rather not manually do this to every object I wish to print that has a hole printed in dead space when it could be bridged for better performance automatically.  Is that clear enough, Mr Science?

bubnikv, 2017-02-15:
This is an evergreen.

On your following picture, you expect Slic3r to draw the red threads.
[image]

What do you expect Slic3r to do in the middle around the hole? Do you expect Slic3r to modify the geometry of the object the same way you did it manually? I think you are asking too much from the slic3r.

Ghrem, 2017-02-15:
I expect nothing in the middle, I expect bridging where it should be but isn't. The last post was an example of forcing the expected behavior.

bubnikv, 2017-02-16:
Maybe in this case your rule makes sense. But there are many cases, which are not that straightforward. For example, what would you propose Slic3r to do, when there is a large overhang, and user does not enable supports?

Ghrem, 2017-02-16:
Does that overhang have 3 sides connected or just 1? When you say overhang I think of just the long side of a line being connected, whereas what I'm showing has 3 connection points, the long side and both ends.  End to end should be bridged since it is over dead space, especially since slice3r tries to print the 3 exterior layer rings in the middle into oblivion first.

DrLex0, 2017-02-21:
That hole is an overhang. The normal way to handle this situation is to print supports. Last time I had to print something similar to this, I added my own support as a thin cylinder underneath the hole, worked like a treat and used the minimum amount of filament.

What you are proposing is to let the slicer print either additional structures that are not present in the 3D model, or omit parts from the model. I would find that unacceptable.

I'm curious about how your proposed method would be supposed to handle this example:
[HoleOverhang.stl.zip](https://github.com/prusa3d/Slic3r/files/788783/HoleOverhang.stl.zip)

bubnikv, 2017-02-21:
@DrLex0 Thanks for an example :-) 

> What you are proposing is to let the slicer print either additional structures that are not present in the 3D model but become part of it (unlike supports), or omit parts from the model. I would find that unacceptable.

I am thinking the same. But I wanted to let @Ghrem find out himself, or to convince me that I am wrong and we can make the world a better place by implementing another heuristics into Slic3r.

closed 2017-02-22 by Ghrem