prusa3d/PrusaSlicer #128: --datadir command line parameter doesn't handle space in path

leavion, 2017-02-14:
### Version
_Version of Slic3r used goes here_
1.33.3-prusa3d-linux64

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Ubuntu 16.04.2 LTS (xenial)
### Behavior
* _Describe the problem_
The --datadir command line parameter doesn't handle spaces in path even when space is escaped or path is surrounded by quote signs.

* _Steps needed to reproduce the problem_
Run slic3r from the command line and pass a path to the datadir which has a space in it.

For example: 

./slic3r --datadir '/home/me/3D Printing/slicer-beta/

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_

Create or use the configuration files in the directory specified in the --datadir parameter and start.

* _Actual Results_
Displays an error:

Input file must have .stl, .obj or .amf(.xml) extension

  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
No. It's a bug report.

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2017-02-14:
@leavion If you edit the slic3r shell script and add the --datadir parameter there, does it work?
I think the problem is in the script, which does not process the parameters with spaces correctly. 
Thanks.

bubnikv, 2017-02-14:
There seems to be two issues:
1) The slic3r is a shell wrapper, calling perl slic3r.pl $@. Here $@ shall be quoted, but it is not.
2) Slic3r only accepts a complete path, it shall understand a relative path or a path starting with ~/

leavion, 2017-02-14:
Indeed, modifying the slic3r script and replacing $@ with "$@" seems to fix the issue.

I now was able to run this and it worked:

`./slic3r --datadir '/home/me/3D Projects/xxx' "/home/me/3D Projects/Prusa/Original-Prusa-i3/Printed-Parts/stl/extruder-cover.stl"`

bubnikv, 2017-02-14:
Cool. I have updated the build script to produce a correct start script.
It would be great to make Slic3r recognize relative paths, but for now I think this is sufficient.
Closing, thanks.

closed 2017-02-14 by bubnikv

leavion, 2017-02-14:
Actually, it will work with relative path and with ~ but you have to escape the spaces and not use quotes around the path to make it work.

For example this works fine once you have "$@" in the script:

`./slic3r --datadir ../../3D\ Projects/xxx ../../3D\ Projects/Prusa/Original-Prusa-i3/Printed-Parts/stl/extruder-cover.st`

as well as this one:

`./slic3r --datadir ~/3D\ Projects/xxx ~/3D\ Projects/Prusa/Original-Prusa-i3/Printed-Parts/stl/extruder-cover.stl`

bubnikv, 2017-02-14:
It will work with escapes, because with escapes the substitution is done by the shell. If you enclose a path into quotes, the substitution will not be done by the shell and it is left to the application. This was actually new to me as well.