prusa3d/PrusaSlicer #129: Slic3r window does not restore after having been minimized on OS X

tcm0116, 2017-02-15:
### Version
1.33.3-prusa3d

### Operating system type + version
OS X 10.11.5

### Behavior
The Slic3r window is not shown when attempting to restore it after having been minimized. To reproduce, click the orange '-' button to minimize the Slic3r window, then click on the Slic3r icon in the dock. The window should restore, but nothing happens. In order to get the window back, right-click on the Slic3r icon in the dock, select "Show All Windows", and then select the Slic3r window that is displayed.

bubnikv, 2017-02-15:
I am not quite familiar with the OSX zoo, but I have tested it quickly on OSX Maveric over SVN and it seems to be working.

tcm0116, 2017-02-15:
I'm running El Capitan. Mavericks is 10.9, so it may behave differently.

bubnikv, 2017-02-17:
I was able to reproduce it using Maverick. I think it is the following issue:

> http://superuser.com/questions/360133/clicking-on-dock-icon-does-not-bring-window-on-top-os-x-lion

I have to talk to bros Prusa. They are both OSX users. I have no idea how the dock thing is supposed to work. The OSX UI is utterly confusing to me, I feel worse than left handed when I have to do something with it.

kelchm, 2017-03-08:
I can confirm this issue in macOS Sierra.

bubnikv, 2017-03-08:
Yes, talked to Michal Prusa, he explained to me how the OSX desktop is supposed to work. So I can confirm this bug.

bubnikv, 2017-03-30:
I am not sure, as I am not an OSX programmer, but there seem to be an API required to make the maximize / minimize working, which is not available through the Perl API, namely 
wxTopLevelWindow::EnableFullScreenView(), wxTopLevelWindow::EnableMaximizeButton() and wxTopLevelWindow::EnableMinimizeButton()

As it is being worked on a port of the Perl UI code to C++, this would once allowed me to call these functions directly from a C++ code. Let's see how the situation will evolve upstream.

mcwienczek, 2021-07-31:
I am having the same problem with MacOS BigSur 11.5.1 and Prusa Slicer 2.3.3

plucker, 2021-11-20:
Same problem with OSX Monterey, the window doesn't restore when you click on the dock icon, it should and 99% of OSX applications behave like this.