prusa3d/PrusaSlicer #130: Sliced Info not displayed

tcm0116, 2017-02-15:
### Version
1.33.3-prusa3d

### Operating system type + version
OS X 10.11.5

### Behavior
No values are displayed in the "Sliced Info" section. I've entered values for Density and Cost in the Filament Settings, re-sliced, removed and re-added the part, and then restarted Slic3r, but still no luck. Is there any other settings that need to be set in order to get this to work?

bubnikv, 2017-02-15:
It is not obvious, but you have to export the g code first to have the
filament statistics calculated. Also you have to fill in the filament
weight and price.

tcm0116, 2017-02-15:
That's definitely not obvious. Is there any way to make it update after it finishes slicing via the Slice Now button? I was experimenting with different settings (infill, layer height, etc.) to see how much difference they would make on the amount of material used.

bubnikv, 2017-02-15:
I would have to look into it deeper. The implementation is by @lordofhyphens, I just pull it into our fork.

lordofhyphens, 2017-02-15:
The information is not available until after gcode generation. 

https://github.com/alexrj/Slic3r/pull/3669

bubnikv, 2017-02-15:
@lordofhyphens I have made the info box to show only when the information is available and valid.
https://github.com/prusa3d/Slic3r/commit/2ddabe5fa8e40519f8a5df4eda7ca8ebd3a265b0

bubnikv, 2017-02-15:
I have edited the title, so the issue is left here as an improvement suggestion.

mischlrebl, 2017-05-08:
+1
it would be nice at least, if the info box would appear even when the export would be cancelled

HansPeterHaastrup, 2018-10-13:
+1 (and +1 for each of the 100 students i've been teaching to use slicer)

I have background processing turned on so the model is automatically sliced when I load it into slicer. also, if I change settings the model will be sliced immediately. I would imagine that this generates a temporary g-code file? 
if yes, wouldn't the sliced info be available through that?

Kadah, 2018-11-07:
Thought this was a bug as sliced info panel was seemingly only showing up at random then noticed today that it would only show up after export. Definitely non-obvious.

This also effects the preview feature type controls/filters, and the feature colors, which change after gcode export.

bubnikv, 2018-12-11:
Automatic generation of the G-code into a temp file & showing the info automatically will be part of the upcoming 1.42 alpha release. Closing.

closed 2018-12-11 by bubnikv