prusa3d/PrusaSlicer #132: Crash on OS X first time slicing with variable layer height due to free/malloc issue

pieterbos, 2017-02-15:
### Version
1.33.4-prusa3d

### Operating system type + version
OS X 10.12.2

### Behavior
Tried the variable layer height. Editing went fine. Then I tried 'slice now' and it crashed with the attached stack trace: 
[slic3r crash report](https://github.com/prusa3d/Slic3r/files/778600/slic3r.crash.report.txt)

It mentions some memory was freed that wasn't allocated, perhaps a memory management, pointer or multithreading issue.

I tried restarting Slic3r and tried again and this time it worked fine.

bubnikv, 2017-02-16:
@pieterbos Thanks Pieter for the report. 

From the log

2   libsystem_c.dylib             	0x00007fffa7577420 abort + 129
3   libsystem_malloc.dylib        	0x00007fffa766703f free + 530
4   Wx.bundle                     	0x000000010340342b XS_Wx__Event_DESTROY(interpreter*, cv*) + 139
5   perl5.22.0                    	0x0000000102c54d53 Perl_pp_entersub + 2659

it seems it is some race condition in the Perl / wxWidgets, so it is likely not specific to the new layer editing tool. But I may be wrong.

Please keep me posted with the crash reports, if it happens again. If it gets serious, I will try to implement some workaround for this particular wxPerl.

Recently the slic3r was behaving quite well on OSX since I started building the Perl & wxWidgets on OSX from sources, making our OSX loving customer happy.

bubnikv, 2017-02-21:
@pieterbos Is this issue reproducible with https://github.com/prusa3d/Slic3r/releases/tag/version_1.33.7
or may I close the issue, please?

pieterbos, 2017-03-07:
I cannot reproduce this problem anymore.

closed 2017-03-07 by pieterbos