prusa3d/PrusaSlicer #133: Сrash at startup 1.33.4 W8

lonelymyp, 2017-02-16:
### Version
1.33.4 x32/64

### Operating system type + version
W8 x64

### Behavior
Сrash at startup If the program is running from the disk D, from disk С run without problems

bubnikv, 2017-02-16:
@lonelymyp What is the full path of your folder, where you unpacked slic3r on the D: drive?
Does it crash if you unpack it directly to d:\ ?

lonelymyp, 2017-02-16:
D:\Slic3r\slic3r.exe - Ok
D:\Users\Фёдор\Downloads\Slic3r-1.33.4-prusa3d-win32-full-201702151702\Slic3r-1.33.4\slic3r.exe - crash

bubnikv, 2017-02-16:
All right, Фёдор, I think this is the issue.

Slic3r is written in C++ & Perl, Perl does not fully support Unicode access (yes, in year 2017) to Windows file system. 

What Windows localization is it? Was it been installed as English, or Russian (maybe Bulgarian :-) ) Windows? Is your name written with the code page, which is default to your Windows installation? Sorry, I know these are very technical questions.

burbilog, 2017-02-16:
Yeah, that's how Russian Windows works. I'd say 95% of Windows installations in Russia are using cyrillic names in paths and that's a huge pain in the ass with many non-unicode-aware programs. Windows installer ask for user's name and people enter their name, even Russian language is pre-selected.

bubnikv, 2017-02-16:
I am Czech, when installing Slic3r into a nicely named folder
D:\temp\ščéířáýěščíéřýáěščé\Slic3r-1.33.4-prusa3d-win64-full-201702151702
it crashes as well. Something to look into.

bubnikv, 2017-02-16:
There seems to be something bad going on with Perl & WxWidgets. I am not sure whether I will be able to fix this issue any time soon.

bubnikv, 2017-02-16:
@lordofhyphens FYI, this is another surprise of wxPerl.

bubnikv, 2017-02-16:
I sent an e-mail to 	wxperl-users@perl.org
but I don't expect this problem to be fixed any time soon. I think my time is better invested working on slicing algorithms and on getting rid of Perl than on fixing an abandoned wxPerl.

CReimer, 2017-02-16:
Getting rid of Perl sounds nice. I already tried to remove most of Build.pl stuff, but Perl doesn't make it easy to build a Perl module outside of their weird Module::Build environment.

bubnikv, 2017-02-16:
It seems the perl itself is broken and the core perl dynaloader fails to work, if the perl was started with long names. For example, running slic3r with the following slic3r.bat fails:

"%~dp0\perl5.24.0.exe" "%~dp0\slic3r.pl" %*

It works though, if one replaces the magic %~dp0 with %~dps0:

"%~dps0\perl5.24.0.exe" "%~dps0\slic3r.pl" %*

That way both the paths of the perl.exe and slic3r.pl use the 8.3 characters only per directory entry.

lordofhyphens, 2017-02-16:
I am working on a C++ port of the GUI for main branch.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

bubnikv, 2017-02-17:
Fixed by executing perl and setting the library paths to short (8.3) paths.
The slic3r.exe has been modified to execute the perl interpreter with 8.3 paths if started from a directory with non-ISO characters, and an information is printed out to a console in such a case. If slic3r is installed in a directory with ISO characters only (and without spaces and semi-colons), the long path is used as before.

closed 2017-02-17 by bubnikv