prusa3d/PrusaSlicer #134: [Request] Add option to specify infill/perimeter overlap for bridges

tcm0116, 2017-02-16:
The part below is hollow, so the first layer after the void is essentially all bridges. As can be seen from the picture, some of the strings didn't attach to the perimeter, so they just fell down into the part. I tried increasing the overall infill/perimeter overlap from 15% to 25%, but that made the bottom layers very messy. Going back own to 20% was a pretty reasonable compromise as the bottom and top layers don't look too bad and most of the bridge strings attached. 

My proposal is to add an option for specifying the infill/perimeter overlap for bridges. This way, I can use 15% for the top and bottom layers and 25% for the bridges to ensure they attach to the perimeter. 

[image]

bubnikv, 2017-02-16:
I was actually looking into a similar issue today. When printing an object over raft in Slic3r, the 1st object layer is printed with bridge flow as any other surface over support. And indeed the infills did not fuse with the perimeters.

So this is a real issue and it needs to be addressed, but I am not sure whether it calls for another parameter.   In my opinion Slic3r should caculate the correct line spacing just from the geometry of the extrusions.

tcm0116, 2017-02-17:
So I've been looking into how to possibly do this. In the standard Slic3r, the infill_overlap is applied after it has been determined if the fill surfaces a bridge (in LayerRegion::make_fill), so it would be realtively easy to just apply a different overlap. However, this logic seems to have been re-worked in the Prusa edition such that the fill surfaces are all initially generated as internal surfaces with the infill_overlap generically applied (in PerimeterGenerator::process), and then the type of surface is determined at a later stage (in LayerRegion::process_external_surfaces). As such, I haven't been able to determine a straight-forward way to increase the infill_overlap for bridges. Any ideas?

bubnikv, 2017-02-18:
@tcm0116 The implementation in Slic3r Prusa Edition is the original implementation of the infill overlap.I know @alexrj has worked on the infills over the Christmas, he accepted my C++ infills and then did some other things to it. I did no have time to analyze his changes, but I guess this is where the logic changed. I will look into his recent changes soon.

The current bridging flow logic is flawed in two ways:

1) The layer placed directly over the bridges leaves gaps (the green areas), which shall be filled in my opinion. Maybe it is not possible to fill these gaps fully due to the bending of the bridge threads, but to some extent the 2nd object layer should be pressed into the bridges, especially if there is a support under the bridges, or the bridges are printed over a raft.

[image]

2) The issue you have raised: Insufficient overlap of the bridges with the bridging perimeters.

[image]

One probably wants to lower the 1st object layer to squish the two layers against each other to fill the voids:

[image]

Now the last image of mine contradicts with your proposal to some extent. If we are to lower the 1st layer over the bridges. this would squish the bridging infill to the bridging perimeters. This only makes sense, if the bridging perimeters are laid over supports (or raft) and they are sufficiently fused to the supports to not be pulled inside by the bridging infills. So actually we may have anther useful requirement, to extrude the bridging perimeters in such a way to fuse them to the supports more firmly than the bridging infill.

This is a complex topic and I don't think a simple hack would be good enough.

bubnikv, 2017-02-18:
Looking into your original comment, what you really want to achieve is an overlap of the bridging infill specifically with the internal NON-bridging perimeter.

I have also an impression, that you printed your object with zero bottom layers, right? This is a very specific use case, normally one would model the object with the bottom hollow, then if the vertical wall was thick enough, Slic3r would anchor the bridging layer into the perimeters correctly.

tcm0116, 2017-02-18:
> you printed your object with zero bottom layers, right?

No. Here's a cross-section view of my part.

[image]

As you can see, supports wouldn't work since I would not have the ability to remove them, so I have to bridge the bottom layer after the void. In this case, the second flaw you raised is the issue I'm having. With respect to the first flaw, squishing the first layer after the bridge would not be an option since there's nothing under the bridge layer to support it while the second layer is being squished down.

In general, it seems to me that any unsupported bridge needs to be firmly attached to the bridging perimeters in order to ensure that they don't just fall down as in my original image. I would think that even supported bridges would want to be fairly attached to the perimeters as well in order to provide a solid foundation for the layers above once the part is finished and the supports are removed.

> I know @alexrj has worked on the infills over the Christmas

I didn't realize that @alexrj had a newer version of the infills. Since it looks relatively simple to make my change to the his current version, perhaps I'll attempt it there to see how well it works.

tcm0116, 2017-02-21:
I'm kind of thinking out loud here, but what if unsupported bridges actually occupied one of the perimeters as in the image below?

[image]

In this case, the bridge wouldn't have to overlap the perimeter since the ends would actually be supported by the perimeter on the layer below. Obviously, this wouldn't work for parts with a single perimeter. I suppose it could perhaps weaken the part at the bridge layer since there would be one less perimeter, but bottom layer bridges should be 100% infill, so the bridge would pretty much replace the perimeter.

What are your thoughts on this idea?

fiveangle, 2017-05-02:
In S3D they use an "extra" parameter for the bridge to simply have the slicer pretend the bridge extends this extra amount, making it independent from infill overlap.

fiveangle, 2017-05-02:
At least I think that's what it's supppsed to do. With it set to 1mm, even super fast bridges stick no problem.

tcm0116, 2017-05-02:
I've leavened a few things done I put this together that might help others. In my case, the wall that was supporting the bridge was the same thickness as my perimeter thickness setting. I have found that if the wall thickness is greater than the perimeter thickness, then that will result in there being a ledge below the bridging layer that the bridges will adhere to very well. In other words, Slic3r will work as in my image above for that example if the number of perimeters was set to 2. If the walls were three perimeters thick, then you could set the number of perimeters to 2, and so on.

Gebhartj, 2019-07-28:
To solve this problem I use a 1% concentric fill

the78mole, 2020-03-09:
I encounter the same problem... 
[image]
[image]
Maybe the model is a bit extreme, having only two perimeters, but taking one off and use it for the bridging infill would make life much easier...

trashcandrummer, 2021-04-18:
I am extremely happy that I have found this issue and that there are more people who have this problem. Printing certain models with PETG is a pain because of this missing feature. Hope that an implementation can be found for that feature and that it will soon find its way into the next release

fredbar, 2023-04-12:
Same here. Bridges do not connect correctly to walls and infils due to not having any overlap. This makes crappy top surfaces and fragile parts.
Please consider this.

See this: it behaves very very poorly.
 
[image]

cringegnere, 2023-04-21:
I agree, a specific overlap for bridge lines over perimeters would it's a must have.

bitingsock, 2023-05-12:
as a potential work around, if your bridge is the only infill in that layer (or if you don't care about the side effects to other infill for that layer) you can use a height modifier and change the settings for that height.
[image]

viesturz, 2023-11-15:
+1 this would be super userful.

gaiuscosades, 2023-12-07:
+1 this is very much needed

In addition IMHO the _infill/perimeter overlap_ parameter must be seperable from a _solid infill/perimeter overlap_ parameter.

In my experience it often works out like this:
1. I have a part with bad bonding on bridges or infill in general so I increase the overlap
 2. I increase the overlap
 3. I get rough top and bottom finishes due to overextrusion next to the perimeters due to excess solid infill not being able to go anywhere
 4. I reduce overlap again until both (suboptimal bonding bridges and rough surface) are tolerable again.
 5. I am somewhat unhappy.

sms8179, 2025-01-31:
+1. Would love to see separate infill/perimeters overlap for internal solid infill vs. sparse infill.

Tupson444, 2026-09-13:
still relevant

SachCZ, 2026-09-30:
Hello, this is clearly a feature request and as such it now belongs to discussions where it can be upvoted. I am moving it there.