prusa3d/PrusaSlicer #135: Compiling with VS2013

wfcook, 2017-02-16:
### Version
09a5e69

### Operating system type + version
Windows 10 x64, VS2013 free

### Behavior
I'm trying to follow the procedure at [https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#building-with-visual-studio-2013](url) and of course it's blowing up because OpenGL 0.6704 is no longer readily available at http://search.cpan.org/~chm/

Is there a more complete or current set of instructions somewhere?

bubnikv, 2017-02-16:
No, there are currently no better instructions. I have attached the old OpenGL package, but you should now be able to compile with the newest OpenGL. If that works, please update the wiki.

[OpenGL-0.6704.tar.gz](https://github.com/prusa3d/Slic3r/files/780906/OpenGL-0.6704.tar.gz)

wfcook, 2017-02-16:
I'll keep plugging at it. The issue is that the new openGL doesn't like the patches for an x64 build, and it's not immediately obvious how to apply them manually.

wfcook, 2017-02-16:
Yeah the issue is with building openGL 0.70 using VS2013 for x64. Might be a PEBCAK error but I have succeeded in building the older library, thanks

bubnikv, 2017-03-23:
We have now new instructions to compile with VS2013 with pre-compiled libraries.
https://github.com/prusa3d/Slic3r/wiki/How-to-compile-Slic3r-Prusa-Edition-on-MS-Windows

closed 2017-03-23 by bubnikv