prusa3d/PrusaSlicer #136: Slic3r 1.33.4 hangup on start

roboprint, 2017-02-16:
### Version
Slic3r-1.33.4-prusa3d-win64-full-201702151702

### Operating system type + version
I'm using 64-bit Win10 version 1511 os build 10586.753, english user interface. Windows username contains only standard latin characters, windows user home directory also does not contain non-latin chars.

### Behavior
Both new releases of Slic3r (1.33.4 and 1.33.3) and both (win32 and win64) versions just hangs up on startup with 100% cpu usage.

I have tried startup with previous version configs in user directory and without configs at all.

The same behavior from all disks/paths, including c:\Slic3r and orginal c:\Slic3r-1.33.4-prusa3d-win64-full-201702151702

[image]

Previous release of Slic3r (1.31.6) working great without any problems.

Call stack of main thread in debugger:
```>	libwinpthread-1.dll!00000000649442c6()	Unknown
 	libwinpthread-1.dll!0000000064945485()	Unknown
 	libgcc_s_sjlj-1.dll!000000006cece3bb()	Unknown
 	libstdc++-6.dll!000000006fcf069d()	Unknown
 	[External Code]	
 	perl524.dll!000000006db69f93()	Unknown
 	perl524.dll!000000006dbeaf6a()	Unknown
 	perl524.dll!000000006dbeb046()	Unknown
 	Wx.xs.dll!00000000703f0ccc()	Unknown
 	wxbase30u_gcc_custom.dll!00000000699c4146()	Unknown
 	wxbase30u_gcc_custom.dll!0000000069be228f()	Unknown
 	wxbase30u_gcc_custom.dll!0000000069be2984()	Unknown
 	wxbase30u_gcc_custom.dll!0000000069be2a0e()	Unknown
 	wxbase30u_gcc_custom.dll!0000000069be2abd()	Unknown
 	wxbase30u_gcc_custom.dll!0000000069be2b18()	Unknown
 	wxbase30u_gcc_custom.dll!0000000069be2782()	Unknown
 	wxmsw30u_core_gcc_custom.dll!0000000064fc967e()	Unknown
 	wxmsw30u_core_gcc_custom.dll!0000000064fcb7ad()	Unknown
 	wxmsw30u_core_gcc_custom.dll!0000000064fb282c()	Unknown
 	[External Code]	
```

OttoES, 2017-02-17:
I can confirm this. Same on my system. Windows 10 Home 64-bit Intel i5-2450M. Language English. Path is c:\3dprinting\Slicerp
Tried different paths and 32-bit and 64-bit versions.
Simply hangs on the opening screen and uses most of the processing time.

bubnikv, 2017-02-17:
That's bad. I suspect the OpenGL drivers. I will try to provide a build, which will initialize the OpenGL shaders first after the Layer Editing button is pressed.

bubnikv, 2017-02-17:
> Both new releases of Slic3r (1.33.4 and 1.33.3) and both (win32 and win64) versions just hangs up on startup with 100% cpu usage.

There is not much to do about it right now. I believe the OpenGL driver hangs somewhere. To make Slic3r usable on computers with buggy OpeGL drivers, I have moved the initialization of OpenGL shaders to the first invocation of the Layer Editing button. I have also enabled the printout of the OpenGL shader compilation error. If the next Slic3r release hangs on you, I am most interested into the result of the shader compilation log, which you may get by running

slic3r-console.exe --gui

https://github.com/prusa3d/Slic3r/commit/89702a5b4e63b783193bfbfcf1bddeeaf404a7c4

bubnikv, 2017-02-17:
I am closing the issue now. Please reopen the issue if you can reproduce the problem with 1.33.5. In that case, please provide a long printed out by slic3r-console.exe

roboprint, 2017-02-17:
@bubnikv thank you very much!

New version (1.33.5) is able to start, but hungs up after I press Layer editing button with the following message in console:

`Can't call method "Load" on an undefined value at C:/Slic3r-1.33.5-prusa3d-win64-full-201702171510/lib/Slic3r/GUI/3DScene.pm line 201.`

bubnikv, 2017-02-17:
@roboprint  would you please provide information on your graphics card, driver, OpenGL capabilities?
http://www.ozone3d.net/gpu_caps_viewer/

roboprint, 2017-02-17:
@bubnikv 
```
===================================[ Graphics Adapters / GPUs ]
- GPU 1
  - Name: Intel(R) G45/G43 Express Chipset (Microsoft Corporation - WDDM 1.1)
  - GPU codename: GMA X4500
  - Revision ID:    3
  - Driver: 8.15.10.2702 (3-11-2013)
  - Bus Id: -1
  - Shader cores: 10
  - BIOS version: Intel Video BIOS
  - Memory size: 3854MB
  - Memory type: 
  - Memory bus width: 0-bit

===================================[ OpenGL GPU Capabilities ]
- OpenGL vendor: Microsoft Corporation
- OpenGL renderer: GDI Generic
- OpenGL Version: 1.1.0
- GLSL (OpenGL Shading Language) Version: n.a.
- ARB Texture Units: 0
- Vertex Shader Texture Units: 0
- Pixel Shader Texture Units: 0
- Geometry Shader Texture Units: 0
- Max Texture Size: 1024x1024
- Max Anisotropic Filtering Value: X1.0
- Max Point Sprite Size: 1.0
- Max Dynamic Lights: 8
- Max Viewport Size: 16384x16384
- Max Vertex Uniform Components: 0
- Max Fragment Uniform Components: 0
- Max Geometry Uniform Components: 0
- Max Varying Float: 0
- Max Vertex Bindable Uniforms: 0
- Max Fragment Bindable Uniforms: 0
- Max Geometry Bindable Uniforms: 0
- Frame Buffer Objects (FBO) Support:[no]
- Multiple Render Targets / Max draw buffers: 0
- Pixel Buffer Objects (PBO) Support:[no]
- S3TC Texture Compression Support:[no]
- ATI 3Dc Texture Compression Support:[no]
- Texture Rectangle Support:[no]
- Floating Point Textures Support:[no]
- OpenGL Extensions: 2 extensions (GL=2 and WGL=0)
	- GL_WIN_swap_hint
	- GL_EXT_bgra
- OpenGL core capabilities: 176 caps listed
	- GL_MAX_LIST_NESTING: 64
	- GL_MAX_EVAL_ORDER: 30
	- GL_MAX_LIGHTS: 8
	- GL_MAX_CLIP_PLANES: 6
	- GL_MAX_TEXTURE_SIZE: 1024
	- GL_MAX_PIXEL_MAP_TABLE: 65536
	- GL_MAX_ATTRIB_STACK_DEPTH: 16
	- GL_MAX_MODELVIEW_STACK_DEPTH: 32
	- GL_MAX_NAME_STACK_DEPTH: 128
	- GL_MAX_PROJECTION_STACK_DEPTH: 10
	- GL_MAX_TEXTURE_STACK_DEPTH: 10
	- GL_MAX_VIEWPORT_DIMS: 16384
	- GL_MAX_CLIENT_ATTRIB_STACK_DEPTH: 16
	- GL_MAX_3D_TEXTURE_SIZE: 0
	- GL_MAX_ELEMENTS_VERTICES: 256
	- GL_MAX_ELEMENTS_INDICES: 2048
	- GL_MAX_TEXTURE_UNITS: 0
	- GL_MAX_CUBE_MAP_TEXTURE_SIZE: 0
	- GL_MAX_TEXTURE_LOD_BIAS: 0
	- GL_MAX_DRAW_BUFFERS: 0
	- GL_MAX_VERTEX_ATTRIBS: 0
	- GL_MAX_TEXTURE_COORDS: 0
	- GL_MAX_TEXTURE_IMAGE_UNITS: 0
[... 130 lines cut ...]
	- GL_MAX_COMPUTE_SHARED_MEMORY_SIZE: 0
	- GL_MAX_COMPUTE_UNIFORM_COMPONENTS: 0
	- GL_MAX_COMPUTE_ATOMIC_COUNTER_BUFFERS: 0
	- GL_MAX_COMPUTE_ATOMIC_COUNTERS: 0
	- GL_MAX_COMBINED_COMPUTE_UNIFORM_COMPONENTS: 0
	- GL_MAX_COMPUTE_WORK_GROUP_INVOCATIONS: 0
	- GL_MAX_VERTEX_ATTRIB_RELATIVE_OFFSET: 0
	- GL_MAX_VERTEX_ATTRIB_BINDINGS: 0
	- GL_MAX_UNIFORM_LOCATIONS: 0
	- GL_MAX_FRAMEBUFFER_WIDTH: 0
	- GL_MAX_FRAMEBUFFER_HEIGHT: 0
	- GL_MAX_FRAMEBUFFER_LAYERS: 0
	- GL_MAX_FRAMEBUFFER_SAMPLES: 0
	- GL_MAX_COMPUTE_VARIABLE_GROUP_INVOCATIONS_ARB: 0
	- GL_MAX_COMPUTE_FIXED_GROUP_INVOCATIONS_ARB: 0
	- GL_MAX_COMPUTE_VARIABLE_GROUP_SIZE_ARB: 0
	- GL_MAX_COMPUTE_FIXED_GROUP_SIZE_ARB: 0
	- GL_MAX_SPARSE_TEXTURE_SIZE_ARB: 0
	- GL_MAX_SPARSE_3D_TEXTURE_SIZE_ARB: 0
	- GL_MAX_SPARSE_ARRAY_TEXTURE_LAYERS_ARB: 0
	- GL_SPARSE_TEXTURE_FULL_ARRAY_CUBE_MIPMAPS_ARB: 0
	- GL_MAX_CULL_DISTANCES: 0
	- GL_MAX_COMBINED_CLIP_AND_CULL_DISTANCES: 0
- OpenGL extension capabilities: 1 caps listed
```

bubnikv, 2017-02-17:
How did you generate the list?

Slic3r relies on the following OpenGL ARB extensions:
ARB_shader_objects
ARB_fragment_shader
ARB_vertex_shader
ARB_shading_language_100

It looks like your driver does not offer any of those. Could you update your graphics driver?

I bet your driver supports the required capabilities, only the OpenGL binding through Perl / wxWidgets supports the OpenGL 1.5 with ARB extensions. While OpenGL 1.5 is supported everywhere on modern computers, the ARB extensions likely not, though these ARB extensions are just another interface to OpenGL 2.0 capabilities. I will likely have to rewrite the OpenGL part to C++ to get access to OpenGL 2.0, but not in this release.

roboprint, 2017-02-17:
> How did you generate the list?

I used program that you provided (gpu caps viewer), last tab contains button that produces this info as far as I remember.

> Could you update your graphics driver?

Sorry, but it's impossible. The only way to upgrade drivers is to buy new hardware. Both Microsot and Intel abandoned driver development for such 'old' Intel G45 chipset. So I'm using default drivers provided by Win10.

bubnikv, 2017-02-17:
I see now. I overlooked, that you driver only supports opengl 1.1. This
looks like the Windows software only renderer. I think I will sleep better
now. There is not much anyone can do with such a graphics card.

I shall integrate some diagnostics though.

roboprint, 2017-02-17:
I understand that. In any case I'm happy that I can at least run and use Slic3r 1.33.5 after your fix. Thank you for your support.

OttoES, 2017-02-17:
Thanks for the quick response.  You are correct, Slic3r will now start up but hangs on Layer Editing. I tried to update my graphics driver but there are no updates. 

GPU Shark v0.9.10
(C)2013 Geeks3D - www.geeks3d.com
----------------------------------------
- Elapsed time: 00:00:34
- Windows 10 64-bit build 14393
- OpenGL info:
 - GL_VERSION: 3.1.0 - Build 9.17.10.4229 (# ext: 129)
 - GL_RENDERER: Intel(R) HD Graphics 3000
  GPU 1 - Intel(R) HD Graphics 3000
- GPU: SandyBridge(GT2) (rev:  9)

- Driver version: 9.17.10.4229
- Bios version: Intel Video BIOS
- GPU memory size: 2076MB
- GPU cores: 12
- GPU Texture units: 1

bubnikv, 2017-02-18:
@OttoES Would you please run slic3r-console.exe from cmd.exe and send me the print-out?

OttoES, 2017-02-18:
I did it as follows: open cmd, run slic3r-console.exe. The gui started up - nothing new on the console. Loaded a stl file - all fine. Pressed "Layer Editing" crash with the following on the console:

c:\3Dprinting\Slicer133\Slic3r>slic3r-console.exe
Info about Body1.STL:
  size:              x=32.111 y=114.000 z=135.261
  number of facets:  4922
  number of shells:  1
  volume:            18944.064
  needed repair:     yes
  degenerate facets: 0
  edges fixed:       34
  facets removed:    0
  facets added:      0
  facets reversed:   0
  backwards edges:   0
Can't call method "Load" on an undefined value at c:/3Dprinting/Slicer133/Slic3r/lib/Slic3r/GUI/3DScene.pm line 201.

Once again thanks for your time and effort. Just a side comment that it does work on my other computer.

bubnikv, 2017-02-18:
@OttoES It seems your OpenGL does not support one of the following legacy OpenGL extensions:

ARB_shader_objects
ARB_fragment_shader
ARB_vertex_shader
ARB_shading_language_100

Would you please send me the following statistics?

[image]

I will likely not be able to fix it, but at least I will be able to do a correct diagnostics on the target computer.

Thanks

OttoES, 2017-02-18:
I really appreciate all the trouble but if this is an isolated problem I will just live with it. Here is the report you asked for. Thanks

===================================================
GPU Caps Viewer v1.32.0.1 report
http://www.geeks3d.com
http://www.ozone3d.net/gpu_caps_viewer/
===================================================

===================================[ System / CPU ]
- CPU Name: Intel(R) Core(TM) i5-2450M CPU @ 2.50GHz
- CPU Core Speed: 2494 MHz
- CPU logical cores: 4
- Family: 6 - Model: 10 - Stepping: 7
- Physical Memory Size: 6144 MB
- Operating System: Windows 10 64-bit build 14393
- PhysX Version: drivers not installed

===================================[ Graphics Adapters / GPUs ]
- Current Display Mode: 1366x768 @ 60 Hz - 32 bpp
- Num GPUs: 1

- GPU 1
  - Name: Intel(R) HD Graphics 3000
  - GPU codename: SandyBridge(GT2)
  - Device ID: 8086- 126
  - Subdevice ID: 17AA-3975
  - Revision ID:    9
  - Driver: 9.17.10.4229 (5-27-2015) - GL:ig4icd64.dll
  - Bus Id: -1
  - Shader cores: 12
  - Texture units: 1
  - TDP: 35W
  - BIOS version: Intel Video BIOS
  - Memory size: 2076MB
  - Memory type: 
  - Memory bus width: 0-bit

===================================[ OpenGL GPU Capabilities ]
- OpenGL vendor: Intel
- OpenGL renderer: Intel(R) HD Graphics 3000
- OpenGL Version: 3.1.0 - Build 9.17.10.4229
- GLSL (OpenGL Shading Language) Version: 1.40 - Intel Build 9.17.10.4229
- ARB Texture Units: 8
- Vertex Shader Texture Units: 16
- Pixel Shader Texture Units: 16
- Geometry Shader Texture Units: 0
- Max Texture Size: 8192x8192
- Max Anisotropic Filtering Value: X16.0
- Max Point Sprite Size: 255.0
- Max Dynamic Lights: 8
- Max Viewport Size: 8192x8192
- Max Vertex Uniform Components: 512
- Max Fragment Uniform Components: 1024
- Max Geometry Uniform Components: 0
- Max Varying Float: 41
- Max Vertex Bindable Uniforms: 0
- Max Fragment Bindable Uniforms: 0
- Max Geometry Bindable Uniforms: 0
- Frame Buffer Objects (FBO) Support:[yes]
- Multiple Render Targets / Max draw buffers: 8
- Pixel Buffer Objects (PBO) Support:[yes]
- S3TC Texture Compression Support:[yes]
- ATI 3Dc Texture Compression Support:[no]
- Texture Rectangle Support:[yes]
- Floating Point Textures Support:[no]
- MSAA: 4X
- OpenGL Extensions: 130 extensions (GL=116 and WGL=14)
	- GL_EXT_blend_minmax
	- GL_EXT_blend_subtract
	- GL_EXT_blend_color
[... 331 lines cut ...]
	- GL_MAX_VERTEX_TEXTURE_IMAGE_UNITS_ARB: 16 (GL_ARB_multitexture)
	- GL_MAX_TEXTURE_IMAGE_UNITS_ARB: 16 (GL_ARB_multitexture)
	- GL_MAX_CUBE_MAP_TEXTURE_SIZE_ARB: 8192 (GL_ARB_texture_cube_map)
	- GL_NUM_COMPRESSED_TEXTURE_FORMATS: 6 (GL_ARB_texture_compression)
	- GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT: 16.000000 (GL_EXT_texture_filter_anisotropic)
	- GL_MAX_VERTEX_UNIFORM_COMPONENTS: 512 (GL_ARB_vertex_shader)
	- GL_MAX_VARYING_FLOATS: 41 (GL_ARB_vertex_shader)
	- GL_MAX_VERTEX_ATTRIBS: 16 (GL_ARB_vertex_shader)
	- GL_MAX_TEXTURE_IMAGE_UNITS: 16 (GL_ARB_vertex_shader)
	- GL_MAX_VERTEX_TEXTURE_IMAGE_UNITS: 16 (GL_ARB_vertex_shader)
	- GL_MAX_COMBINED_TEXTURE_IMAGE_UNITS: 16 (GL_ARB_vertex_shader)
	- GL_MAX_TEXTURE_COORDS: 8 (GL_ARB_vertex_shader)
	- GL_MAX_FRAGMENT_UNIFORM_COMPONENTS_ARB: 1024 (GL_ARB_fragment_shader)
	- GL_MAX_VERTEX_ATTRIBS_ARB: 16 (GL_ARB_vertex_program)
	- GL_MAX_PROGRAM_MATRICES_ARB: 8 (GL_ARB_vertex_program)
	- GL_MAX_PROGRAM_MATRIX_STACK_DEPTH_ARB: 2 (GL_ARB_vertex_program)
	- GL_MAX_TEXTURE_COORDS_ARB: 8 (GL_ARB_fragment_program)
	- GL_MAX_FRAGMENT_UNIFORM_COMPONENTS: 1024 (GL_ARB_shading_language_100)
	- GL_MAX_GEOMETRY_UNIFORM_COMPONENTS_EXT: 0 (GL_ARB_shading_language_100)
	- GL_MAX_DRAW_BUFFERS_ARB: 8 (GL_ARB_draw_buffers)
	- GL_MAX_COLOR_ATTACHMENTS: 8 (GL_ARB_framebuffer_object)
	- GL_MAX_RENDERBUFFER_SIZE: 4096 (GL_ARB_framebuffer_object)
	- GL_MAX_SAMPLES: 4 (GL_ARB_framebuffer_object)
	- GL_POINT_SIZE_MIN_ARB: 0.000000 (GL_ARB_point_parameters)
	- GL_POINT_SIZE_MAX_ARB: 255.000000 (GL_ARB_point_parameters)

bubnikv, 2017-02-18:
@OttoES It really helps me immensely, if I get a reasonable information from someone, who has having issues with the software. 

Would you please do one more test for me?

Would you please replace a file
Slic3r\lib\Slic3r\GUI\GLShader.pm
with the following:
[GLShader.zip](https://github.com/prusa3d/Slic3r/files/784996/GLShader.zip)

and then run slic3r-console.exe --gui , select the Layer Editing and send me, what Slic3r wrote into the console? Added some debugging information into the Perl module I have attached here. Thanks.

OttoES, 2017-02-18:
I will help you as much as I can. Did the same as before. First without --gui and repeated it with --gui with the same  result:

c:\3Dprinting\Slicer133\Slic3r>slic3r-console.exe
Info about Body1.STL:
  size:              x=32.111 y=114.000 z=135.261
  number of facets:  4922
  number of shells:  1
  volume:            18944.064
  needed repair:     yes
  degenerate facets: 0
  edges fixed:       34
  facets removed:    0
  facets added:      0
  facets reversed:   0
  backwards edges:   0
GL version: 1.1.0, vendor: Microsoft Corporation, renderer: GDI Generic, GLSL version: , ARB:
Missing OpenGL extension: GL_ARB_shader_objects
Missing OpenGL extension: GL_ARB_fragment_shader
Missing OpenGL extension: GL_ARB_vertex_shader
Missing OpenGL extension: GL_ARB_shading_language_100
Shader version: 0
glCreateShaderObjectARB is not supported by this renderer at c:/3Dprinting/Slicer133/Slic3r/lib/Slic3r/GUI/GLShader.pm line 82.

bubnikv, 2017-02-18:
@OttoES this looks like a soft renderer has been instantiated, though you
are having a recent opengl installed.

Aren't you using slic3r over remote desktop?

Would any opengl 2.0 application run? I think an Adobe mesh mixer is a good
example, it will complain, if only opengl 1.x is available.

OttoES, 2017-02-18:
I am not running it over a remote desktop. I will try Adobe mesh mixer - must just download and install it.

OttoES, 2017-02-18:
I installed mesh maker and it runs without any warnings. It runs quit smooth so I would guess that it does use OpenGL or at least some hardware assistance to do the rendering

bubnikv, 2017-02-18:
@OttoES 
Would you please try to start Slic3r with the following executable. Just copy it into the Slic3r directory and run.
[slic3r-console-no-manifest.zip](https://github.com/prusa3d/Slic3r/files/785244/slic3r-console-no-manifest.zip)

It will be rendered in a Windows 95 style without the manifest, but it may initialize the OpenGL driver correctly. Please send me the console log.

It may be very likely the following issue:
http://forum.drdteam.org/viewtopic.php?f=25&t=6696
where the old OpenGL driver refuses to work on Windows 10.

I am still looking on the web for another reasons why Slic3r does not get the right OpeGL context.
Thanks

bubnikv, 2017-02-18:
@OttoES 
You may also try to run Slic3r in a Compatibility mode for some older Windows version.

OttoES, 2017-02-18:
I tried it in both win 8 and win 7 compatibility mode but both crashed. Once again thanks for all the trouble but I think it is just one of those incompatibility things that we will have to live with.

bubnikv, 2017-02-19:
Implemented detection of OpenGL capabilities:
1) Do not crash on "Layer Editing" button, but report the missing OpenGL capabilities. Also detect the software only renderer on Windows.
https://github.com/prusa3d/Slic3r/commit/898deb48c4ade5f4fb25799543fd085daa8378e1
2) New "System Info" menu item and dialog to print out all available information on Slic3r, Perl and OpenGL. https://github.com/prusa3d/Slic3r/commit/70229be9bc2de7c393044256842e94b9beea34bf